gnuradio/gnuradio #7269: grc-qt: AttributeError

dl1ksv, 2024-04-15:
### What happened?

Double clicking a connection line should remove the connection ( or do nothing ? )

### System Information

Linux, F39 build from source

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Start grc --qt

Build a simple flowgraph 
audio source -> audio sink
The double click on the connection line  and you get the err

### Relevant log output

```shell
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib64/python3.12/site-packages/gnuradio/grc/gui_qt/components/canvas/connection.py", line 163, in mouseDoubleClickEvent
    self.parent.connections.remove(self)
    ^^^^^^^^^^^
AttributeError: 'GUIConnection' object has no attribute 'parent'. Did you mean: 'paint'?
```

dl1ksv, 2024-04-15:
In grc --gtk double clicking a connection does the same as a simple click, just selecting the line.

willcode, 2024-04-15:
Needs a deref to the core object.

closed 2024-04-21 by willcode