gnuradio/gnuradio #7270: grc-qt: Fails to build example with bus ports

dl1ksv, 2024-04-15:
### What happened?

Opening an example using bus ports in grc --qt results in an core dump

### System Information

Linux F39, build from scratch

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Start

grc --qt

Select

File -> Examples -> fec -> ber_curve_gen.grc

### Relevant log output

```shell
QGraphicsScene::removeItem: item 0x55f4ce08cca0's scene (0x0) is different from this scene (0x55f4cdf302a0)
QGraphicsScene::removeItem: item 0x55f4ce017b20's scene (0x0) is different from this scene (0x55f4cdf302a0)
QGraphicsScene::removeItem: item 0x55f4ce073190's scene (0x0) is different from this scene (0x55f4cdf302a0)
QGraphicsScene::removeItem: item 0x55f4ce03c160's scene (0x0) is different from this scene (0x55f4cdf302a0)
QGraphicsScene::removeItem: item 0x55f4ce097fd0's scene (0x0) is different from this scene (0x55f4cdf302a0)
QGraphicsScene::removeItem: item 0x55f4ce087f40's scene (0x0) is different from this scene (0x55f4cdf302a0)
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib64/python3.12/site-packages/gnuradio/grc/gui_qt/components/canvas/port.py", line 81, in itemChange
    self.update_connections()
  File "/usr/local/gnuradio/lib64/python3.12/site-packages/gnuradio/grc/gui_qt/components/canvas/port.py", line 77, in update_connections
    conn.gui.set_rotation(self.parentItem().rotation())
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'rotation'
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib64/python3.12/site-packages/gnuradio/grc/gui_qt/components/canvas/port.py", line 81, in itemChange
    self.update_connections()
  File "/usr/local/gnuradio/lib64/python3.12/site-packages/gnuradio/grc/gui_qt/components/canvas/port.py", line 77, in update_connections
    conn.gui.set_rotation(self.parentItem().rotation())
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'rotation'
Abgebrochen (Speicherabzug geschrieben)
```

willcode, 2024-04-15:
Probably need a operate on the gui object, not the core object.

dl1ksv, 2024-04-29:
Just as a reminder: Even if this issue will be fixed, this example and some other examples will remain brocken.
See #5198