gnuradio/gnuradio #7276: top_block thread safety -  is top_block_impl::wait() call thread safe ?

FromHub, 2024-04-19:
### What happened?

While reconfiguring flowgraph with lock/unlock and having another dedicated thread blocked on wait, application crash occurs after a while (many lock/unlock operations). This crash is notified by glibc with message such as 'double free or corruption (fasttop)' after unlock.
Detailed design : grc based application (python + PyQt). lock/unlock called by the GUI thread, wait called by an additional thread (QThread).

An attached debugger gives the following trace (partial) :
  - gr::top_block_impl::wait()
  - gr::thread::thread_group::join_all()
  - boost::thread::join_noexcept()
  - [...]
  - __GI___libc_free

top_block_impl::wait calls top_block_impl::wait_for_jobs() and access d_scheduler without protection.

    void top_block_impl::wait_for_jobs()
    {
        if (d_scheduler)
            d_scheduler->wait();
    }

The occurrence of the crash seems to indicate a threading problem caused by a gnuradio framework issue.
Note : this scenario is legal as in python wait is called on a dedicated thread through _top_block_waiter.run function (python/gnuradio/gr/top_block.py)

The hypothesis is that scheduler is sometimes destroyed a second time by wait_for_jobs causing the application to crash.
Maybe a fix could be limited to use the scheduler through a shared_ptr to postpone the destruction of the 'old' scheduler when wait_for_jobs has ended.

    void top_block_impl::wait_for_jobs()
    {
        scheduler_sptr scheduler(d_scheduler)
        if (scheduler)
            scheduler->wait();
    }

Proposal : make top_block::wait thread safe.

### System Information

OS: Ubuntu 22.04.4 LTS
GR Installation Method: apt

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.7.0

### Steps to Reproduce the Problem

The hypothesis is a threading issue so dependent of thread execution. One should verify the code with a review to confirm the problem or instrument the code to force the issue occurrence.

The following code tries to mimic the behavior outside gnuradio framework .

/* Includes */
#include <unistd.h>     /* Symbolic Constants */
#include <sys/types.h>  /* Primitive System Data Types */ 
#include <errno.h>      /* Errors */
#include <stdio.h>      /* Input/Output */
#include <stdlib.h>     /* General Utilities */
#include <pthread.h>    /* POSIX Threads */
#include <string.h>     /* String handling */
#include <memory>
#include <vector>
#include <cstdio>
#include <unistd.h>

struct Scheduler {
    Scheduler(int id):id(id) {}

    int id;
    
    void wait()
    {
        usleep(2000000);
        printf("using scheduler %d\n", id);
    }
    
    ~Scheduler()
    {
        printf("destroying scheduler %d\n", id);
    }
};

struct top_block
{
    top_block():scheduler(new Scheduler(1))
    {
    }
    
    void wait()
    {
        if( scheduler)
            scheduler->wait();  
    }
    
    void wait_fix()
    {
        std::shared_ptr<Scheduler> ptr(scheduler);
        if( ptr)
            ptr->wait();
    }
    
    void unlock()
    {
        usleep(1000000);
        printf("changing scheduler %u\n", scheduler->id);
        std::shared_ptr<Scheduler> ptr(new Scheduler(2));
        scheduler = ptr;
        printf("new scheduler %u\n", scheduler->id);
    }
    
    std::shared_ptr<Scheduler> scheduler;
};

void* call_unlock ( void *ptr )
{
    top_block* block =  (top_block *) ptr;
    
    block->unlock();
    
    pthread_exit(0);
}

void* call_wait ( void *ptr )
{
    top_block* block =  (top_block *) ptr;
    
    block->wait(); // or block->wait_fix();
    
    pthread_exit(0);
}

int main()
{
    pthread_t thread1, thread2;
    
    top_block block;
    
    pthread_create (&thread1, NULL, &call_unlock, (void *) &block);
    pthread_create (&thread2, NULL, &call_wait,   (void *) &block);

    pthread_join(thread1, NULL);
    pthread_join(thread2, NULL);

    exit(0);
}

marcusmueller, 2024-04-19:
Hi!

> The hypothesis is that scheduler is sometimes destroyed a second time by wait_for_jobs causing the application to crash. Maybe a fix could be limited to use the scheduler through a shared_ptr to postpone the destruction of the 'old' scheduler when wait_for_jobs has ended.

Hm, that hypothesis seems wrong: `d_scheduler` is a member of `top_block_impl` and does *not* get destructed at the end of `wait_for_jobs`, at all. (The object it points to *might* get destroyed when in `unlock()` the method `restart` is called, which replaces `d_scheduler` with a different shared pointer to a scheduler; but that's a pretty orderly process: the reference counter in the old `d_scheduler` shared pointer is reduced, and only if that was the last place holding a copy of that shared_ptr, the underlying object gets destroyed; unless you're simultaneously from two different threads are modifying top_block_impl, this means there's always an object to access.)

`d_scheduler` is already a `shared_ptr<scheduler>` and the pointed-to `scheduler` only gets destructed once; I'd call that pretty safe, because these shared pointers are inherently thread-safe, that's why they are relatively expensive.

> The occurrence of the crash seems to indicate a threading problem caused by a gnuradio framework issue.

I agree, but my guess is that the real problem here is that you're modifying `top_block` from two separate threads. If you don't do that, does the problem still occur? 

Would it be possible to share an actual reproduction source code of the problem? Your illustrative code in your issue report is cool, but it only demonstrates the problem if your hypotheses are correct, and figuring that out would be the first problem to tackle.

FromHub, 2024-04-20:
Hi Marcus,

Thanks for your reply and the details. Reading back your answer I notice that I mix the debugger trace and the behavior of the issue simulator. The issue might be caused by the access to an object that have already been destroyed and not a double destruction.
The second thread doing monitoring through the wait call access the scheduler without a proper ownership on the scheduler. Because GNU radio framework is used on multi-core architecture, the monitoring thread notices the scheduler end (stop) before a new scheduler is build and the old one destroyed. However this is not guaranty and a nasty crash occurs when this precondition is not respected : the monitoring thread uses the old scheduler that have already been destroyed. It exists only one scheduler_sptr for 2 threads without protections ! To me it is not intentional top_block_impl::wait being not thread safe.
I have not plan to provide a GNU radio based reproduction code (without effort it is too unpredictable to get a reproducible immediate scenario).

argilo, 2025-02-03:
I believe I found the root cause while looking into a CI failure; see https://github.com/gnuradio/gnuradio/issues/7551#issuecomment-2629815218 for details.