gnuradio/gnuradio #7280: Creating C++ OOT with gr-modtool guide leads to segmentation fault.

Phoenix-64, 2024-04-19:
### What happened?

I tried following [this](https://wiki.gnuradio.org/index.php?title=Creating_C%2B%2B_OOT_with_gr-modtool) guide on creating my own C++ module. 
But I get a segmentation fault:
```
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x00007ffff183ae65 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.7
```

### System Information

OS: WSL2 Ubuntu 22.04
GR Installation Method: apt installed.
Using VS Code.
Building with:  GCC 11.4.0 x86_64-linux-gnu

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.7.0

### Steps to Reproduce the Problem

Run attached Test_gen.py file:
`sudo python3 Test_gen.py`
To use Test_gen.grc compile attached c++ and header file.
Then generate flowgraph from Test_gen.grc
Execute

[example_files.tar.gz](https://github.com/gnuradio/gnuradio/files/15045397/example_files.tar.gz)

### Relevant log output

```shell
=thread-group-added,id="i1"
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Warning: Debuggee TargetArchitecture not detected, assuming x86_64.
=cmd-param-changed,param="pagination",value="off"
Stopped due to shared library event (no libraries added or removed)
Loaded '/lib64/ld-linux-x86-64.so.2'. Symbols loaded.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff0847640 (LWP 25658)]
[New Thread 0x7ffff0046640 (LWP 25659)]
[New Thread 0x7fffef845640 (LWP 25660)]
[New Thread 0x7fffef044640 (LWP 25661)]
[New Thread 0x7fffee843640 (LWP 25662)]
[New Thread 0x7fffee042640 (LWP 25663)]
[New Thread 0x7fffed841640 (LWP 25664)]
[New Thread 0x7fffed040640 (LWP 25665)]
[New Thread 0x7fffec83f640 (LWP 25666)]
[New Thread 0x7fffec03e640 (LWP 25667)]
[New Thread 0x7fffe8c5c640 (LWP 25670)]
[... 199 lines cut ...]
Loaded '/lib/x86_64-linux-gnu/libxcb-shm.so.0'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxcb-keysyms.so.1'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxcb-randr.so.0'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxcb-render-util.so.0'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxcb-render.so.0'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxcb-shape.so.0'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxcb-sync.so.1'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxcb-xfixes.so.0'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxcb-xinerama.so.0'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxcb-xkb.so.1'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxcb-xinput.so.0'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libSM.so.6'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libICE.so.6'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxkbcommon-x11.so.0'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxkbcommon.so.0'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libxcb-util.so.1'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libXcursor.so.1'. Symbols loaded.
Loaded '/lib/x86_64-linux-gnu/libXfixes.so.3'. Symbols loaded.
Loaded '/usr/lib/x86_64-linux-gnu/qt5/plugins/platforminputcontexts/libcomposeplatforminputcontextplugin.so'. Symbols loaded.
Warning: Source file '/home/benjamin/GNU_learning/gr-customModule/lib/multDivSelect_impl.cc' is newer than module file '/usr/local/lib/x86_64-linux-gnu/libgnuradio-customModule.so.1.0.0'.
Execute debugger commands using "-exec <command>", for example "-exec info registers" will list registers in use (when GDB is the debugger)

Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x00007ffff183ae65 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.7
```

Phoenix-64, 2024-04-19:
A Test file containing just a signal generator throttle and time sink works without problems

marcusmueller, 2024-04-19:
Hi! Could you actually roll an archive of the full OOT? This just looks like a bug somewhere in your code, but I can't test it because I can't build it locally, missing the rest of the OOT.

If it helps, we know that C++-containing OOTs work just fine, so this is really unlikely this is a problem with the generated code.

Phoenix-64, 2024-04-20:
Yes, I am pretty sure it is a problem in my code. However, I do not know if I made the mistake or if there is a problem with the tutorial. Thanks for looking at it
[full_oot.tar.gz](https://github.com/gnuradio/gnuradio/files/15049178/full_oot.tar.gz)

> Hi! Could you actually roll an archive of the full OOT? This just looks like a bug somewhere in your code, but I can't test it because I can't build it locally, missing the rest of the OOT.
> 
> If it helps, we know that C++-containing OOTs work just fine, so this is really unlikely this is a problem with the generated code.
> 
> Also, since you already seem to have that in GDB: cool! Could you run `backtrace`? that would show us what is segfaulting, where. (you would need to run another compile and install step, because GDB complains that you've changed the source code from what you've installed.

Phoenix-64, 2024-04-20:
I seem to be missing something. Because I also tried the [following ](https://wiki.gnuradio.org/index.php/Guided_Tutorial_GNU_Radio_in_C%2B%2B)tutorial to no avail. I also get a Segmentation fault error.

marcusmueller, 2024-04-22:
thanks, will look into it later today

Phoenix-64, 2024-04-26:
Okay solved, I think I was missing the qt5 dev lib. And I needed to add the FindQtw.cmake file to discover it.

closed 2024-04-26 by Phoenix-64