gnuradio/gnuradio #7284: grc-qt: a double clicking a connection deletes the connection but the fg stays valid

dl1ksv, 2024-04-21:
### What happened?

Take a simple fg

signal source(float) -> audio sink

Double clicking the connection removes the connection but the fg stays valid and can even be generated. No errors are displayed and the fg is still marked as saved , so you can close the fg without further actions. The fg should be marked as changed and invalid.

Taking the same fg:
After a simple click on the connection you can delete the connection by the delete button. You can delete the connection by clicking the delete button. The fg is marked as invalid. Errors are displayed. Closing the fg requires either to discard or to save the changes.

### System Information

Linux, F39, build from scratch

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

See description above

willcode, 2024-04-21:
This is after #7278, right?

dl1ksv, 2024-04-21:
Correct

PR #7287 (2024-05-10): grc-qt: Fixing incorrect error display

## Description

Fixes #7284

In addition:

If on startup grc loads an invalid flowgraph, the fg is shown as valid, that is the error button is disabled.

If there are several tabs ar open that contain a mixture of valid and invalid fg's the error button is not adopted to the status of the selected tab.

This pr calculates  the status of the buttons individually for each tab depending on it's fg.

## Related Issue

Fixes #7284
## Testing Done
closed 2024-05-10 by willcode