gnuradio/gnuradio #7291: gnuradio-companion --qt filename.grc doesn't open filename.grc

marcusmueller, 2024-04-23:
### What happened?

Tried to open a GRC flowgraph directly with QT GRC; it doesn't do that, doesn't throw an error. The same file works when opened via menu

### System Information

OS: linux
GR Installation Method: source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-716-g77155087

### Steps to Reproduce the Problem

gnuradio-companion --qt filename.grc

dl1ksv, 2024-04-24:
At the moment grc --qt does not care about filename.grc.
But should grc --qt add filename.grc to the list of recently opened files  or only open filename.grc?

marcusmueller, 2024-04-24:
Both, I'd say: This is what I want to happen when people doubleclick a GRC file in their file manager.

dl1ksv, 2024-04-24:
Is grc now behaveing as expected?

PR #7294 (2024-05-08): grc-qt: grc --qt filename.grc doesn't open filename.grc

## Description

Now the list of files specified on the cl for grc is appended to the list of files to be opened on startup.

fixes #7291

## Checklist

- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significant change per commit. 
- [x] I [have signed my commits before making this PR](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md#dco-signed)
- [ ] My code follows the code style of this project. See [GREP1.md](http [... cut]
commit 3df1fe09 (2024-05-08): grc-qt: grc --qt filename.grc doesn't open filename.grc

Now the list of files specified on the cl for grc is appended to
the list of files to be opened on startup.

fixes #7291
commit 7c7cc1bb (2024-05-17): grc-qt: grc --qt filename.grc doesn't open filename.grc

Now the list of files specified on the cl for grc is appended to
the list of files to be opened on startup.

fixes #7291

(cherry picked from commit 3df1fe091e538525397b243b5107b84043070e92)
closed 2024-05-08 by willcode