gnuradio/gnuradio #7295: gnuradio-companion --qt missing oot module causes SIGABRT in python

chrisjohgorman, 2024-04-30:
### What happened?

I got a sigabrt from python when trying to load a grc file.  
I believe this is caused by the fact that my install was missing an out of tree module referenced in the grc file.  Installing the missing oot module caused the error to disappear.

```
[chris@descartes ~]$ Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/gnuradio/grc/gui_qt/components/canvas/port.py", line 81, in itemChange
    self.update_connections()
  File "/usr/lib/python3.12/site-packages/gnuradio/grc/gui_qt/components/canvas/port.py", line 77, in update_connections
    conn.gui.set_rotation(self.parentItem().rotation())
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'rotation'
```
The following is the core file backtrace.
```
Core was generated by `/usr/bin/python3 /usr/bin/gnuradio-companion --qt'.
Program terminated with signal SIGABRT, Aborted.
#0  0x0000715f3eaab32c in ?? () from /usr/lib/libc.so.6
[Current thread is 1 (Thread 0x715f3f287740 (LWP 989090))]
(gdb) bt
#0  0x0000715f3eaab32c in ??? () at /usr/lib/libc.so.6
#1  0x0000715f3ea5a6c8 in raise () at /usr/lib/libc.so.6
#2  0x0000715f3ea424b8 in abort () at /usr/lib/libc.so.6
#3  0x0000715efda980c2 in QMessageLogger::fatal(char const*, ...) const ()
    at /usr/lib/libQt5Core.so.5
#4  0x0000715efe048048 in ??? ()
    at /usr/lib/python3.12/site-packages/PyQt5/QtCore.abi3.so
#5  0x0000715f01a3fe75 in ??? ()
    at /usr/lib/python3.12/site-packages/PyQt5/sip.cpython-312-x86_64-linux-gnu.so
#6  0x0000715efd08d837 in ??? ()
    at /usr/lib/python3.12/site-packages/PyQt5/QtWidgets.abi3.so
#7  0x0000715efd230975 in ??? ()
    at /usr/lib/python3.12/site-packages/PyQt5/QtWidgets.abi3.so
#8  0x0000715efcc8fb30 in ??? () at /usr/lib/libQt5Widgets.so.5
#9  0x0000715efcca9868 in QGraphicsView::paintEvent(QPaintEvent*) ()
    at /usr/lib/libQt5Widgets.so.5
#10 0x0000715efd1e9724 in ??? ()
    at /usr/lib/python3.12/site-packages/PyQt5/QtWidgets.abi3.so
#11 0x0000715efc98e040 in QWidget::event(QEvent*) ()
    at /usr/lib/libQt5Widgets.so.5
#12 0x0000715efca3ca83 in QFrame::event(QEvent*) ()
    at /usr/lib/libQt5Widgets.so.5
--Type <RET> for more, q to quit, c to continue without paging-- 
#13 0x0000715efd1ea5ec in ??? ()
    at /usr/lib/python3.12/site-packages/PyQt5/QtWidgets.abi3.so
#14 0x0000715efdc90a42 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#15 0x0000715efc9578ef in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () at /usr/lib/libQt5Widgets.so.5
#16 0x0000715efd2e3b57 in ??? ()
    at /usr/lib/python3.12/site-packages/PyQt5/QtWidgets.abi3.so
#17 0x0000715efdc93a68 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () at /usr/lib/libQt5Core.so.5
#18 0x0000715efc981f7b in QWidgetPrivate::sendPaintEvent(QRegion const&) ()
    at /usr/lib/libQt5Widgets.so.5
#19 0x0000715efc98337d in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) () at /usr/lib/libQt5Widgets.so.5
#20 0x0000715efc983d8e in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) ()
    at /usr/lib/libQt5Widgets.so.5
#21 0x0000715efc98281e in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) () at /usr/lib/libQt5Widgets.so.5
#22 0x0000715efc983d8e in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, --Type <RET> for more, q to quit, c to continue without paging--
QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) ()
    at /usr/lib/libQt5Widgets.so.5
#23 0x0000715efc98281e in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) () at /usr/lib/libQt5Widgets.so.5
#24 0x0000715efc983d8e in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) ()
    at /usr/lib/libQt5Widgets.so.5
#25 0x0000715efc983bec in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) ()
    at /usr/lib/libQt5Widgets.so.5
#26 0x0000715efc98281e in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) () at /usr/lib/libQt5Widgets.so.5
#27 0x0000715efc983d8e in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) ()
    at /usr/lib/libQt5Widgets.so.5
[... 53 lines cut ...]
    at /usr/lib/libpython3.12.so.1.0
#62 0x0000715f3ee6cc67 in Py_RunMain () at /usr/lib/libpython3.12.so.1.0
#63 0x0000715f3ee28fab in Py_BytesMain () at /usr/lib/libpython3.12.so.1.0
#64 0x0000715f3ea43cd0 in ??? () at /usr/lib/libc.so.6
#65 0x0000715f3ea43d8a in __libc_start_main () at /usr/lib/libc.so.6
#66 0x0000619c6fe33045 in _start ()
```

### System Information

OS: Arch Linux
Gnu Radio installation, built from source for 3.10.10.0

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

1) Run gnuradio-companion --qt
2) Attempt to load a grc file with out of tree module blocks in it that are not currently installed.  (For example install gr-satellites, build a flowgraph with one of the satellite blocks, then remove gr-satellites and try to load the grc file you created.)

haakov, 2024-05-08:
Can reproduce. This error occurs when a flowgraph has missing blocks that are connected to other blocks. Fix coming shortly

haakov, 2024-05-08:
@chrisjohgorman Does https://github.com/gnuradio/gnuradio/pull/7329 fix the issue?

chrisjohgorman, 2024-05-08:
@haakov #7329 fixes it for me.   Thanks.

chrisjohgorman, 2024-05-08:
@haakov, apparently my brain is broken today.  Should I close this as you've fixed it?

haakov, 2024-05-08:
No, we can keep it open until the fix has been merged :)

closed 2024-05-09 by willcode