gnuradio/gnuradio #7298: grc-qt block variables with long names exit the block enclosure

chrisjohgorman, 2024-05-01:
### What happened?

Under the gtk front end, the blocks do not have writing that exits their boundaries.  The qt blocks do if the variable name is long.  I've attached a screenshot of showing this with a signal source block.

[image]

### System Information

OS: Arch Linux
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

1) Add either a lowpass filter block or a signal source block to a flowgraph.  The writing for window.WIN_HAMMING or  analog.GR_COS_WAVE exits the block.

chrisjohgorman, 2024-05-01:
I see I may have been a little overzealous here.  Disabling `Show parameter expressions in block` fixes this for the most part.  I don't know whether or not the blocks should resize when this is enabled, perhaps this problem can be put to the bottom of the pile or closed.

[image]

marcusmueller, 2024-12-31:
I think this is fixed by now; making a value long "stretches" the block.

closed 2024-12-31 by marcusmueller