gnuradio/gnuradio #7299: grc-qt variables do not evaluate their python values

chrisjohgorman, 2024-05-01:
### What happened?

Put a variable on the canvas and set it's value to `48e3*4*8`.  The gtk grc evaluates this variable and shows it as 1.536M.  Currently the grc-qt interface leaves the value as `48e3*4*8`.

[image]

### System Information

OS: Arch Linux
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

1) place a variable block on the screen.
2) set it's value to a python expression.  eg. 100*100
3) see that the expression is not evaluated, ie. it shows 100*100 rather than 10000

willcode, 2024-07-17:
Reproduced. Note that it is just the display that is different than the Gtk version. The values are still evaluated correctly for runtime. I kind of like this way (display of the expression) better, actually.

marcusmueller, 2024-12-31:
You can actually select whether the display shows computed or raw values. But regardless of whether "show expression in block" or "show value in block" is selected in the preferences, the expression is always shown.