gnuradio/gnuradio #7300: grc-qt block library search path does not clear itself with a second ctrl+F

chrisjohgorman, 2024-05-01:
### What happened?

Searching through the blocks via a second call to ctrl+F, does not clear the text entry box and reset the blocks.  In the attached picture, ctrl+F was used to add a Qt Gui Range to the canvas.  A second ctrl+F was executed and the result can be seen in the attached image. 
1)  The text entry box is returned to where the user stopped typing, rather than the beginning of the textbox and the remainder of the previous block name is still present in the text entry box.
2)  The library of modules in the table is not reset back to it's original list view.

[image]

The second picture is of the original state, which I believe a second call to ctrl+F should yield.

[image]

### System Information

OS: Arch Linux
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

1) use ctrl+F to find a block
2) put that block on the canvas
3) use ctrl+F a second time to find a different block
4) the text entry box must be manually cleared to proceed with the second search.

haakov, 2024-05-05:
Note that you can add blocks by hitting Enter! 

So if you want to add a null source, you can hit Ctrl+F, write "null so" and hit Enter.

haakov, 2024-05-05:
@chrisjohgorman Does https://github.com/gnuradio/gnuradio/pull/7315 fix the issue?

chrisjohgorman, 2024-05-05:
@haakov #7315 does fix the issue for me.  Thank you for the fix.

Chris

closed 2024-05-08 by willcode