gnuradio/gnuradio #7301: grc-qt File/Save does not automatically append a .grc extension to the file

chrisjohgorman, 2024-05-01:
### What happened?

When saving a file the file/save code does not append a .grc extension to the saved file.  This causes a second issue, where the file saved cannot be found under linux as the search is for a file named `*.*`.  
Here is a picture of the first issue.
[image]
This causes the second issue, where the file cannot be found even when searching for `*.*` as the filename has no `.` character in it.
[image]
The file saved in the first issue is not visible in the list of files, even though it was just saved.

### System Information

OS: Arch Linux
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

1) File/Save a filename without any extension.
2) Try to load that file from File/Open.

dkozel, 2024-05-01:
@chrisjohgorman Thanks for taking the time to file all the GRC Qt reports! They'll help a lot in bringing it up to (and past) parity with the GTK version.

chrisjohgorman, 2024-05-01:
@dkozel You're very welcome for the bug reports.  I've been keen to try out the new interface ever since I found out about it's availability in 3.10.10.0.  I'm sure the new interface will be in parity with the GTK version soon enough.  I have one more bug to report, but that will have to wait until tomorrow.  

One thing I do like about the QT interface is that it is quite quick in comparison to the GTK interface.

PR #7312 (2024-05-08): grc-qt: Save flowgraph: add extension to filename

## Description

At the moment you can save a flowgraph without the extension grc.

This pr adds the extension if missing.

Fixes: #7301
## Related Issue

See #7301 
## Testing Done

Saving and reopening a fg with and without extension grc
## Checklist
commit 2742beb1 (2024-05-08): grc-qt: Save flowgraph add extension to filename

At the moment you can save a flowgraph without the extension grc.

This pr adds the extension if missing.

Fixes: #7301
commit 5059fef1 (2024-05-17): grc-qt: Save flowgraph add extension to filename

At the moment you can save a flowgraph without the extension grc.

This pr adds the extension if missing.

Fixes: #7301

(cherry picked from commit 2742beb1e1901a344e77b76883f16142d1ef4127)
closed 2024-05-08 by willcode