gnuradio/gnuradio #7305: grc-qt Variable Editor window does not reinitialize when a new flowgraph is slected

chrisjohgorman, 2024-05-03:
### What happened?

A new flowgraph fails to refresh the variable editor window and populate the window with the variables from the new flowgraph.

### System Information

OS: Arch Linux
GR Installation Method: Source

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

1) Open a flowgraph
2) Select the `new` button
3) The new flowgraph is displayed, but in the Variable Editor window the variables are from the old flowgraph.

chrisjohgorman, 2024-05-03:
I just noticed an error in this bug report.  

Gnu Radio Version should be 3.10 (maint-3.10)

haakov, 2024-05-19:
Looks like this has been fixed with https://github.com/gnuradio/gnuradio/pull/7287

PR #7357 (2024-05-21): grc-qt: Variable editor: fix several bugs

## Description

The Variable editor in grc-qt is an incomplete port of the Variable editor in grc --gtk

But as all the functionsof the editor like  add/delete/enable/disable block are already coded int the Mainwindow it's sufficient to map this functions from the editor by signal and slot mechanism to the correspondend functions in the Mainwindow.
## Related Issue

Fixes #7305
Fixes #7306
Fixes #7307

## Testing Done
commit f360dce2 (2024-05-21): grc-qt: Variable editor: fix several bugs

The Variable editor in grc-qt is an incomplete port of the Variable editor in grc --gtk

But as all the functionsof the editor like  add/delete/enable/disable block are
already coded int the Mainwindow it's sufficient to map this functions from the editor
by signal and slot mechanism to the correspondend functions in the Mainwindow.

Fixes: #7305
Fixes: #7306
Fixes: #7307
commit 05055ac1 (2024-05-27): grc-qt: Variable editor: fix several bugs

The Variable editor in grc-qt is an incomplete port of the Variable editor in grc --gtk

But as all the functionsof the editor like  add/delete/enable/disable block are
already coded int the Mainwindow it's sufficient to map this functions from the editor
by signal and slot mechanism to the correspondend functions in the Mainwindow.

Fixes: #7305
Fixes: #7306
Fixes: #7307

(cherry picked from commit f360dce2bff8ce8797ba1bb6feb9c0b70b703704)
closed 2024-05-21 by willcode