gnuradio/gnuradio #7310: `gr_modtool help` should consistently output what `gr_modtool --help` does

marcusmueller, 2024-05-04:
### Feature Description

s. title.

### Feature Urgency

low (just an idea)

ajitg25, 2024-05-05:
Can you elaborate more on this?

sb424dat, 2024-05-06:
What is the difference in output?

SWITCHin2, 2024-07-06:
@marcusmueller I would like to work on it, could assign it to me and also any reference to follow?

marcusmueller, 2024-07-07:
@SWITCHin2  it's all yours to work on!

Reference to follow: Work against the `main` branch.

Stick with the general style of the code. Code formatting ideally done using pycodestyle or flake8; see the `.flake8` file in the gnuradio/ directory.

Python needs to be clean, generally, but we're not overly picky. Code needs to work with Python 3.7.2 ; when you open a PR, you'll see that we run CI on various platforms.

marcusmueller, 2024-07-21:
@SWITCHin2 have you had any success?

SWITCHin2, 2024-07-22:
@marcusmueller not yet!! was trying understand the flow but getting a bit confused, will try my best to fix it soon

SWITCHin2, 2024-07-22:
Any possible area to look upon?  would be helpful

marcusmueller, 2024-07-23:
@SWITCHin2 in gnuradio/gr-utils/modtool/cli you find all the implemented commands. Look for example at info.py. You can copy that file as `help.py`, and start changing it to do what it should!

SWITCHin2, 2024-07-23:
Ok! Thanks :) So did this means currently there is no file like that for `help` command?I was actually for that.

marcusmueller, 2024-08-04:
Hi! What's the state of things, @SWITCHin2

hariharandakshnamurthy, 2024-11-20:
@marcusmueller  is this issue still open can i work on it?

marcusmueller, 2024-11-20:
It says "Open", right :) So, yes, please go ahead!
And: if you have any problems, ask! (It's better to make progress and ask, then to be stuck and not ask :) ). I'm currently heading to bed, but there should be someone online most of the day on our chat, which is on matrix, [#gnuradio:gnuradio.org](https://app.element.io/#/room/#gnuradio:gnuradio.org).

kmr-ankitt, 2024-12-14:
Can i work over this issue?

marcusmueller, 2024-12-14:
@kmr-ankitt yes, as I told literally anyone else in here: any issue that's "open" is ready to be fixed :) Go ahead :) Looking forward to your pull request. See the comment above about how you can reach out to the community in case you need any help!

PR #7573 (2024-12-30): modtool: added gr_modtool help command that should work exactly like gr_modtool --help
## Description

This pull request introduces a new `gr_modtool help` command that functions equivalently to `gr_modtool --help`. This addition improves consistency and user experience by integrating the help functionality into the `gr_modtool` CLI in a more intuitive way. The implementation uses the `click` library and the `subprocess` module to execute the existing `gr_modtool --help` command and display its output.

This change resolves issue #7310.

---

## Related Issue

Fixes #7310.

---

## Which block [... cut]
closed 2024-12-30 by marcusmueller