gnuradio/gnuradio #7313: GRC bug when generating processing a graph with RFNoC TX Streamer

ptrkrysik, 2024-05-05:
### What happened?

Any graph that contains RFNoC Tx Streamer block causes following error:
```
Generate Error: Undefined
>>> Failure
Traceback (most recent call last):
  File "/home/user/prefix/lib/python3.11/site-packages/gnuradio/grc/gui/Application.py", line 762, in _handle_action
    generator.write()
  File "/home/user/prefix/lib/python3.11/site-packages/gnuradio/grc/core/generator/top_block.py", line 86, in write
    for filename, data in self._build_python_code_from_template():
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/prefix/lib/python3.11/site-packages/gnuradio/grc/core/generator/top_block.py", line 135, in _build_python_code_from_template
    connections=self._connections(),
                ^^^^^^^^^^^^^^^^^^^
  File "/home/user/prefix/lib/python3.11/site-packages/gnuradio/grc/core/generator/top_block.py", line 328, in _connections
    code = template.render(
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/mako/template.py", line 439, in render
    return runtime._render(self, self.callable_, args, data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 874, in _render
    _render_context(
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 916, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 943, in _exec_template
    callable_(context, *args, **kwargs)
  File "memory:0x7c3000154350", line 32, in render_body
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 230, in __str__
    raise NameError("Undefined")
NameError: Undefined

```
Checked on the head of maint-3.10 and main branches. Both have this issue.

### System Information

OS: Ubuntu 20.04
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-717-g72e21b54

### Steps to Reproduce the Problem

Create a graph with 'RFNoC Tx Streamer' block.
For example 'Signal Source' -> 'RFNoC Tx Streamer' -> 'RFNoC TX Radio':
[rfnoc_tx_streamer_test.grc.zip](https://github.com/gnuradio/gnuradio/files/15212537/rfnoc_tx_streamer_test.grc.zip)
Try to translate it into Python with:
grcc <graph.grc>

ptrkrysik, 2024-05-05:
After changing default value from 'Auto' to 'Complex int16' (the only other available option) the graph is compiled correctly.

So the real issue is that auto-detection of connection type doesn't work in this case.

polhenarejos, 2024-05-07:
> After changing default value of 'Wire Format' in 'RFNoC Tx Streamer' block properties from 'Auto' to 'Complex int16' (the only other available option), the graph is compiled correctly.
> 
> So the real issue is that auto-detection of connection type doesn't work in this case.

This worked to me.

willcode, 2024-07-17:
@mbr0wn was this fixed by your recent changes?

mbr0wn, 2025-01-28:
Yes, this was fixed by 068049fecce3592907c179fab6e78aa3078a03c4.

closed 2025-01-28 by mbr0wn