gnuradio/gnuradio #7322: freedesktop warning during cmake (GR3.10)

Ech0xing, 2024-05-06:
### What happened?

A bug happened!
CMake Warning (dev) at grc/scripts/freedesktop/CMakeLists.txt:13 (configure_file)

After compiling from source, the icon does not show up (not a PATH problem) and unable to type in gnuradio-companion from the CLI.

### System Information

OS: Ubuntu 20.04.6
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10

### Steps to Reproduce the Problem

Not sure why this is happening, I have installed all of the dependencies. I looked at line 13 from the CMakeList.txt file and cant make heads or tails of it. I have reloaded the laptop twice and still get the same error. This issue causes where I cant find the icon to click on GNURadio Companion under the App menu, nor can i type in gnuradio-companion from the CLI to open.

xdg-utils is installed

git clone https://github.com/gnuradio/gnuradio.git
cd gnuradio
git checkout maint-3.10
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3 ../

This is where part where the Dev error comes up:

-- Configuring JSON/YAML config blocks support...
--   Dependency ENABLE_GRC = ON
--   Dependency JSONSCHEMA_FOUND = TRUE
--   Enabling JSON/YAML config blocks support.
--   Override with -DENABLE_JSONYAML_BLOCKS=ON/OFF
CMake Warning (dev) at grc/scripts/freedesktop/CMakeLists.txt:13 (configure_file):
  configure_file called with unknown argument(s):

   FILE_PERMISSIONS
   OWNER_READ
   OWNER_WRITE
   OWNER_EXECUTE
   GROUP_READ
   GROUP_EXECUTE
   WORLD_READ
   WORLD_EXECUTE

Attached is a screenshot of the CMakeList.txt file that is in question. The highlighted portion is the line in question.
[image]

### Relevant log output

```shell
-- Configuring JSON/YAML config blocks support...
--   Dependency ENABLE_GRC = ON
--   Dependency JSONSCHEMA_FOUND = TRUE
--   Enabling JSON/YAML config blocks support.
--   Override with -DENABLE_JSONYAML_BLOCKS=ON/OFF
CMake Warning (dev) at grc/scripts/freedesktop/CMakeLists.txt:13 (configure_file):
  configure_file called with unknown argument(s):

   FILE_PERMISSIONS
   OWNER_READ
   OWNER_WRITE
   OWNER_EXECUTE
   GROUP_READ
   GROUP_EXECUTE
   WORLD_READ
   WORLD_EXECUTE
```

marcusmueller, 2024-05-07:
Ah, yes, the ``FILE_PERMISSIONS`  flag was added in CMake 3.20, and we only require 3.8. This is a bug.

marcusmueller, 2024-05-07:
exists in grc/scripts/freedesktop/CMakeLists.txt and gnuradio-runtime/python/gnuradio/ctrlport/CMakeLists.txt

Ech0xing, 2024-05-07:
Anything I can do in the meantime to be able to compile GR?

Ech0xing, 2024-05-07:
I figured it out. I fixed the issue in the CMakeList.txt file that fixes the PERMISSIONS command. I will submit a PR see if it make sense.

marcusmueller, 2024-05-08:
> Anything I can do in the meantime to be able to compile GR?

That's just a warning; it should not have stopped you from compiling!