gnuradio/gnuradio #7323: Bump CMake requirement to 3.22 on main

marcusmueller, 2024-05-07:
### Feature Description

As seen in #7294, we tend to use cmake features not present in the fairly ancient minimum CMake version we demand; in that case, we'd need at least 3.20.

Considering cmake versions:

| distro | release | cmake ver |
|--|--|--|
|debian  | oldstable (bullseye)| 3.18| 
|debian | stable (bookworm) |3.25|
|Ubuntu| 22.04 | 3.22|
|Ubuntu| 23.10 | 3.27|
|Ubuntu| 24.04 | 3.28|
|Ubuntu| 24.10| 3.27|
|Fedora | 39 | 3.27|
|Fedora | 40| 3.28 |

I'd say that even for maint-3.10, cmake 3.18 would be safe (and I think even without #7294 we do use 3.16-ish features, anyway) . For main, we should go to at least 3.22 (if we decide that we want to target Ubuntu 22.04 with maint-3.10), or staright for 3.25.

So, this boils down to deciding:

- maint-3.10: Update at all? Updated to 3.18 or 3.22?
- main: Update to 3.22 or 3.25?

### Feature Urgency

medium (would be nice to have in the near future)

dkozel, 2024-05-08:
@johnwparent and @mwoehlke-kitware Tagging you both as an FYI of the discussion and to ask for input on what features you're hoping or expecting to use in the installer work.

Thanks!

willcode, 2024-05-21:
@marcusmueller Ubuntu 20.04 has 3.16 and has EOL in 2025, so we'd have to drop support.

marcusmueller, 2024-06-11:
@willcode  ah sorry for the late reply: I'm not too worried about main development on Ubuntu 20.04 – people using Ubuntu 20.04 should very likely use the nicely maintained 3.10 release series ;)

willcode, 2024-06-11:
I still think of `main` as `maint-3.10-dev`, since we don't know if there will every be a 3.11. Non-backported PRs are effectively orphaned on `main` and can potentially cause conflicts.

marcusmueller, 2024-06-12:
that's more than fine and logical, but that would still mean that for forward development of the maint-3.10 branch we at some point need to drop the Ubuntus we supported on the day 3.10 was released, right. That would arguably be the point where we need to actually do a 3.11… but I mean, speculation.

How about this: We first of all raise the CMake version to 3.16; effectively, we are currently shipping something that I suspect doesn't build with CMake 3.8, anyways, and 3.16 would be a fair representation of the technical base we have a chance of testing against.

willcode, 2024-06-12:
Sure, 3.16 is fine for now.. And I'm not opposed to dropping 20.04 before LTS expires, since we don't tie our fate to Ubuntu or RHEL or anyone else automatically. Just has to be intentional.

marcusmueller, 2024-06-12:
@willcode ha! We were declaring dependency on 3.16 already… we simply weren't checking it. Makes the PR easier.