gnuradio/gnuradio #7327: ImportError  Failed to initialize GTK. If you are running over ssh, did you enable X forwarding and start ssh with -X?  (Typelib file for namespace 'GObject', version '2.0' not found)

rkarikari, 2024-05-08:
### What happened?

A bug happened!

### System Information

OS:  Mac
GR Installation Method:  brew ( Homebrew )

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2_3

### Steps to Reproduce the Problem

In terminal, type ...

`gnuradio-companion`

### Relevant log output

```shell
ImportError

Failed to initialize GTK. If you are running over ssh, did you enable X forwarding and start ssh with -X?

(Typelib file for namespace 'GObject', version '2.0' not found)
```

marcusmueller, 2024-05-08:
Try 

    brew link glib

(from https://github.com/virt-manager/virt-manager/issues/649#issuecomment-2027557677)

If that fixes the issue, this is a brew bug, not a GNU Radio issue, and I'd ask you to report that with the brew folks, because I don't think we can be of much help :)

rkarikari, 2024-05-08:
It did fix the issue.
Thank you.

closed 2024-05-08 by rkarikari

citizenrich, 2024-05-15:
Just for others coming to this issue, I had already installed glib and it was linked, so I did a relink using `brew unlink glib && brew link glib` and gnuradio-companion successfully launched.

marcusmueller, 2024-05-16:
Not at all a brew user myself, but someone did tell the brew recipe maintainers about this, right?