gnuradio/gnuradio #7331: gr_modtool bind not working with Ubuntu 24

mormj, 2024-05-08:
### What happened?

Fresh install of GR (apt install gnuradio-dev) on Ubuntu 24

Created an OOT with gr_modtool, added some blocks, then went to bind the blocks.

Throws a host of GCC errors.  My guess is incompatibility between pygccxml and gcc-13

### System Information

Ubuntu 24

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2

### Steps to Reproduce the Problem

gr_modtool bind

MarkMoss, 2024-05-26:
The attached log shows the errors resulting from running `gr_modtool bind` as the last of the following commands:

- gr_modtool newmod howto
- cd gr-howto
- gr_modtool add -t general -l cpp square_ff
- gr_modtool bind

[gr_modtool bind errors.txt](https://github.com/gnuradio/gnuradio/files/15448383/gr_modtool.bind.errors.txt)

This is with GCC 13.2.0 and pygccxml 2.4.0 (via Ubuntu package python3-pyggxml 2.4.0-1).

marcusmueller, 2024-05-26:
Duplicate of #6477

closed 2024-05-27 by marcusmueller

mbr0wn, 2024-05-27:
Closing this as dupe, amended info in other bug report.