gnuradio/gnuradio #7343: grc-qt pointer arrow does not point into rotated blocks

chrisjohgorman, 2024-05-16:
### What happened?

The arrow pointer that points into a block does not point into blocks properly when they are rotated. 

### System Information

OS: Arch Linux
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

1) Create a flowgraph with a block that is rotated so it is perpendicular to other blocks in the flowgraph.
2) The connection arrow that points into the rotated block doesn't point into the block instead it points in the direction of the block before rotation or out of the block that the connection is going into.
[image]
[image]

marcusmueller, 2024-05-16:
Nice, thanks for the report! And all the other high-quality reports, too, by the way!

chrisjohgorman, 2024-05-16:
You're very welcome Marcus.  I can report them.  I don't know how easy it will be to fix them.

marcusmueller, 2024-05-16:
For a lot of them, the workload is really finding them, taking note of them (if you sit down and say "I'll improve GRC today" without notes of what you need to tackle, you'll not be likely to fix the things that people care about, too easy to lose oversight over what is outstanding), prioritizing, and then sitting down and fixing them.

The "fixing" part isn't necessarily the hardest nor the most time-intense one there! That's why contributors that write bug reports, with explanations, screenshots if applicable, and an honest assessment, are so valuable. Sure, we got limited developer time (both in terms of paid contracts, which were really important in getting grc-qt to where it is now, but have not historically covered the majority of work on it!, and in terms of volunteer developer "after hours hours"), but having time and not knowing what to fix is worse than that :)

willcode, 2024-05-16:
I take it back. This fix was before 3.10.10.0.