gnuradio/gnuradio #7344: QT_GUI ToggleSwitch Dtypes

IsaiahHarvi, 2024-05-16:
### What happened?

Turning on toggle switch to enable a python block sending data to virtual sink

### System Information

OS: PopOS
GR Installation Method: apt

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.1.1

### Steps to Reproduce the Problem

Running an embedded python block to a virtual sink

### Relevant log output

```shell
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/qtgui/toggleswitch.py", line 143, in paintEvent
    painter.drawEllipse(
TypeError: arguments did not match any overloaded call:
  drawEllipse(self, r: QRectF): argument 1 has unexpected type 'float'
  drawEllipse(self, r: QRect): argument 1 has unexpected type 'float'
  drawEllipse(self, x: int, y: int, w: int, h: int): argument 1 has unexpected type 'float'
  drawEllipse(self, center: Union[QPointF, QPoint], rx: float, ry: float): argument 1 has unexpected type 'float'
  drawEllipse(self, center: QPoint, rx: int, ry: int): argument 1 has unexpected type 'float'
```

IsaiahHarvi, 2024-05-16:
I was able to resolve this by type casting parameters to the **drawEllipse** calls inside of the **paintEvent** method

marcusmueller, 2024-05-16:
@IsaiahHarvi  thanks for the report!

marcusmueller, 2024-05-16:
huh, interestingly, it works when I run it from the command line from which I'm also starting GRC-qt, but not from GRC-Qt through execute (the button is just dead).

marcusmueller, 2024-05-16:
ah no wonder I can't reproduce – this was fixed by Volker Schroer in 3.10.2.0, in #5541 .

closed 2024-05-16 by marcusmueller