gnuradio/gnuradio #7347: grc-qt selecting input and output of the same block creates upstream connection causing one upstream block error

chrisjohgorman, 2024-05-17:
### What happened?

I selected the input and output of a throttle block connected in a flowgraph.  This adds a connection from the block preceding the throttle block in the flowgraph to the throttle block.  This connection can be deleted by selecting the arrow going into the throttle block and hitting delete.

I expected that no operation would be performed.

### System Information

OS: Arch Linux
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

1) make a small flowgraph of three blocks, constant source, throttle and null sink.
2) connect all the blocks (constant source -> throttle -> null sink).
3) click on the input of the throttle and the output of the throttle block.
There should now be two connections to the input of the throttle from the constant source.
Here is a picture of the flowgraph now.
[image]
This is the error shown by the errors button.
[image]

willcode, 2024-05-18:
@chrisjohgorman Are the summary and "steps to reproduce" describing the same problem?

chrisjohgorman, 2024-05-19:
Hi @willcode Sorry for the delay in replying.  I just noticed this in my inbox this morning.  The summary and "steps to reproduce" should be describing the same problem.  Basically I found that if had a small three element flowgraph, I could make two connections from the first block in the flowgraph to the second block by following the steps to reproduce.  This appropriately causes the error message 'Domain "stream" can only have one upstream block'.  Perhaps I should have described this more clearly in the summary.  Does this explanation more clearly describe the issue?

dl1ksv, 2024-07-17:
This issue should be fixed with #7402

closed 2024-07-17 by willcode