gnuradio/gnuradio #7353: gr-uhd: is_back_edge argument is ignored when connecting UHD blocks.

pamin98, 2024-05-18:
### What happened?

According to the UHD documentation, graphs consisting of UHD blocks are allowed to have loops if one of the connections is declared a back edge. However, when creating such a design, a runtime error occurs, regardless of whether the boolean is_back_edge argument is set to True. By inspecting the code, one can find the connect() function implementation in gnuradio/gr-uhd/lib/rfnoc_graph_impl.cc:
```
    void connect(const std::string& src_block_id, 
                 const size_t src_block_port,
                 const std::string& dst_block_id,
                 const size_t dst_block_port,
                 const bool is_back_edge) override
    {
        d_logger->debug("Connecting {:s}:{:d} -> {:s}:{:d}",
                        src_block_id,
                        src_block_port,
                        dst_block_id,
                        dst_block_port);
        if (_tx_streamers.count(src_block_id)) {
            if (_rx_streamers.count(dst_block_id)) {
                throw std::runtime_error("Cannot connect RFNoC streamers directly!");
            }
            _graph->connect(_tx_streamers.at(src_block_id),
                            src_block_port,
                            block_id_t(dst_block_id),
                            dst_block_port,
                            _get_adapter_id(src_block_id, src_block_port));
            return;
        }
        if (_rx_streamers.count(dst_block_id)) {
            _graph->connect(src_block_id,
                            src_block_port,
                            _rx_streamers.at(dst_block_id),
                            dst_block_port,
                            _get_adapter_id(dst_block_id, dst_block_port));
            return;
        }

        _graph->connect(block_id_t(src_block_id),
                        src_block_port,
                        block_id_t(dst_block_id),
                        dst_block_port);
    } 
```

To fix the bug, the final call to _graph->connect() should be modified to include the is_back_edge argument, as it is otherwise set by default to False:
```
_graph->connect(block_id_t(src_block_id),
                            src_block_port,
                            block_id_t(dst_block_id),
                            dst_block_port,
                            is_back_edge);
```

### System Information

OS: Ubuntu 
GR Installation Method: Source

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Create a simple design containing a loop, using UHD blocks. Generate and run the code.

pamin98, 2024-07-06:
Is there any update for this issue? Can I open a PR myself to resolve it?

willcode, 2024-07-07:
Yes, you can open a PR, and request review by gmeyerho and mbr0wn.

closed 2025-02-03 by willcode