gnuradio/gnuradio #7356: A function to search blocks in a flowgraph by ID.

bernerus, 2024-05-20:
### Feature Description

There is already one find function for blocks. However this function just looks for blocks in the library.
I want a function with which I can navigate my flowgraph, which has become very large over the years.
I'm looking for the typical find function that now exists on almost any application these days, to find things in my document.
The result could either just select the found blocks or give me a list upon which I can choose to zoom in to the position in the flowgraph where a block is located, or  allow me just to open the block as if I had found it and double-clicked on it.

### Feature Urgency

medium (would be nice to have in the near future)

marcusmueller, 2024-05-20:
Hi @bernerus! Is this about the "old" GRC, or the Qt-port of GRC? we're currently tracking feature requests and bugs for both, so that's why I'm asking :)

haakov, 2024-05-21:
Coincidentally, a PR for this feature already exists (for both Qt and Gtk):
https://github.com/gnuradio/gnuradio/pull/7211