gnuradio/gnuradio #7360: Function Probe is broken due to wrong order of code

nemanjan00, 2024-05-21:
### What happened?

I used Signal Probe and Probe Function blocks. 

Generated python code fails to run, due to `object has no attribute` error

### System Information

OS: Arch Linux
GR Installation Method: Native distro package (3.10.10.0-2)

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

Use both of mentioned blocks

Generated code:

gist.github.com/nemanjan00/058d42e4a878c9f3a91e78461663f8af

willcode, 2024-05-21:
@haakov @dl1ksv if you get a chance, this is a race condition in the python generated by GRC.

willcode, 2024-05-22:
Additional notes in #7360 about potential solutions.

PR #7362 (2024-06-14): Python template startup event
## Description
Add a `self.flowgraph_started` event to the Python Mako code and set it after the top block `start()` is called. This lets anything else that should not run until after everything is initiazed wait on the event.

The code generator does not know that certain blocks (e.g., function probe) depend on other blocks, and especially their accessors, to be available at the time a block may start its own thread. This caused a race between access in a thread spawned by code from a yml template and existance of the `self.` object.

Change the Function Probe Py [... cut]
closed 2024-06-14 by willcode