gnuradio/gnuradio #7370: Can't use Rician on Fading Model block

mas-oliveira, 2024-05-27:
### What happened?

I was trying to use Rician Fading Model to see the influence on some network KPMs but using the gnuradio-companion (GUI) even if I insert a Rician Factor I have an error.

[image]

### System Information

OS: Ubuntu 22.04.6 LTS
GR Installation Method: apt

### GNU Radio Version

3.8  (maint-3.8)

### Specific Version

3.8.1.0

### Steps to Reproduce the Problem

Just add a Fading Model block on the GUI and when selecting Rician/LOS the Rician Factor may not appear and if it appears even if you put a value it would be impossible to start.

pstoeferle, 2024-06-19:
This was fixed in gnuradio 3.10 in commit fbf2082, but did not make it to 3.8 and 3.9  
The issue persists also in all the other blocks, that were changed in this commit.

marcusmueller, 2024-06-23:
Problem is that we avoid having the bandwidth of doing releases for the 3.8 series; so this will probably remain unfixed, unless someone tries to become the release manager for the very-legacy 3.8 series and do these backports and then roll a release, test it, support it…

In all honesty, @mas-oliveira, I think the easiest way here is simply finding the yaml files fixed in the commit that pstoeferle points to, and apply the same fix to them. A better solution for you would be to update your Ubuntu to 24.04LTS, and benefit from all the bugfixes that happened in the meantime :)

closed 2024-06-23 by marcusmueller