gnuradio/gnuradio #7375: Include BLock to control TDD controller in the next update

patilraj1707, 2024-05-29:
### Feature Description

Including the TDD controller block will make the implementation of radars on pluto sdr and other devices easy as it will help to synchronize the TX and RX signals easily.

### Feature Urgency

high (impacts broad portion of ecosystem)

patilraj1707, 2024-05-30:
https://github.com/Yamakaja/gnuradio/tree/feature/gr-iio-tdd
Above is the link to the forked repository but for some reason when I install this gnuradio the flow graphs don't work. They throw a error saying
/home/deadshot/Downloads/TDD.py:11: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.version import StrictVersion
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

>>> Done (return code -11)

marcusmueller, 2024-05-30:
That warning has nothing to do with the problem that the blocks aren't working.
However, knowing that @Yamakaja (hi!) worked on very specific high-speed designs, and different chips than just the AD936x, this hinges on the presence of the specific TDD controller block instantiation on the FPGA of the PlutoSDR; I'd be a bit surprised if that softcore is present, but can't rule it out.

Can you confirm that the TDD controller is present in the stock images for the PlutoSDR?

Yamakaja, 2024-05-30:
Hi! I don't really have time to look into this right now, but I can say that the ad963x HDL contains a TDD instance to control rx/tx TDD feature of the transceiver, but with new (stock!) firmware versions there is apparently an additional TDD instance in the pluto firmware ( https://github.com/analogdevicesinc/hdl/blob/main/projects/pluto/system_bd.tcl#L325-L332 ) which can be used aswell (In this case it is intended for use with the [CN0566 "PHASER"](https://wiki.analog.com/resources/eval/user-guides/circuits-from-the-lab/cn0566), but of course that doesn't mean that it couldn't be used for other purposes ;) )

marcusmueller, 2024-05-30:
Thanks for dropping in, @Yamakaja, and bringing your insight!

@patilraj1707 : I take it you've made that TDD core work for a radar application? If @Yamakaja currently has no time to invest into upstreaming that block, I'll be extremely honest, it'll not be trivial to do it ourselves, especially not in the timeframe of "next update". Can you try to make the block work for your use case, and make a pull-request for that, @patilraj1707 ? GNU Radio (as a project) is very open to contributions, but we need someone to do them :) The hard part might also be writing a block documentation wiki page that explains exactly how to use that block, without which maintenance will be hard – nobody from the recurring contributors, to the best of my knowledge, has your PlutoSDR-TDD-Radar use case, so we wouldn't know when that block is working or not.

patilraj1707, 2024-05-31:
@marcusmueller : I haven't got theh block to work, I wanted to use it for a radar application, I

marcusmueller, 2024-05-31:
@patilraj1707 In that case, what you ask for is a major development effort, not just an "inclusion". We'll keep the feature request around, but realistically, as described above, currently, nobody is well-equipped to help you with that.

marcusmueller, 2024-06-13:
Closing as asked for in #Development

closed 2024-06-13 by marcusmueller