gnuradio/gnuradio #7378: Support NumPy 2.0

jakirkham, 2024-05-31:
### Feature Description

Hi All 👋

NumPy is planning to release 2.0 on June 16th ( https://github.com/numpy/numpy/issues/24300#issuecomment-2125643448 ). NumPy 2.0.0 RC packages for conda & wheels came out at the beginning of April ( https://github.com/numpy/numpy/issues/24300#issuecomment-2030603395 )

NumPy has put out [a migration guide]( https://numpy.org/devdocs/numpy_2_0_migration_guide.html#numpy-2-migration-guide ). More details are in [the release notes]( https://numpy.org/devdocs/release/2.0.0-notes.html )

If GNU Radio makes use of NumPy's C API (and produces wheels that use it), having a release of GNU Radio with wheels built against NumPy 2.0.0rc1 would be helpful to ensure NumPy 1 & 2 compatible wheels (as wheels built against NumPy 1 won't be compatible with NumPy 2). More details in [this NumPy 2 ABI doc]( https://numpy.org/devdocs/dev/depending_on_numpy.html#numpy-2-abi-handling )

Also as NumPy is tracking ecosystem support for NumPy 2.0, it would be helpful to share GNU Radio current support status in issue (with any plans): https://github.com/numpy/numpy/issues/26191

### Feature Urgency

high (impacts broad portion of ecosystem)

ryanvolz, 2024-05-31:
Based on the error seen in the conda-forge build, the first (and hopefully only) fix would seem to be changing [this line](https://github.com/gnuradio/gnuradio/blob/751df659a36aa3b2b4b88040481254640406327b/gr-filter/python/filter/gui/polezero_plot.py#L14) so that `float64` is used instead of `float_`.

marcusmueller, 2024-06-02:
@jakirkham we're using numpy in various places: especially, we have very pervasive binding of C++ vectors of complex floating point values to numpy arrays, mostly done through pybind11.

According to your table, the earliest version of Pybind11 to support numpy2 binding is 2.12.0; that's in no currently released Linux distro.

Reducing to medium, as we currently *can't* upgrade to that due to lack of platform availability, but should still keep track of doing that. The way I see it, even of the few unstable Linux distros that already carry pybind11 2.12, none carry numpy2, so that we have nothing to test against.

jakirkham, 2024-06-03:
Interesting is there a way to separate build dependencies from runtime dependencies?

Would think Pybind11 is only needed at build time. After which it is unneeded for runtime. So users needed not install it from their Linux distro

That said, no very little about GNU Radio is distributed. So very likely am missing important details

As to having NumPy 2 compatibility, generally if packages are built against NumPy 2, they will support NumPy 1 & 2. So that is what is being recommended to users

marcusmueller, 2024-06-03:
> That said, no very little about GNU Radio is distributed. So very likely am missing important details

Mostly, distro packages. We strive, very much, to not be one of these projects that only work in dedicated environments.

> Would think Pybind11 is only needed at build time. 

That is true – but it doesn't mean we can just go ahead and vendor arbitrary versions of it; compatibility hell awaits us, because we have a rich out-of-tree module ecosystem, and can't easily force them to use the same version as us. So, realistically, unless we build software incompatible to the rest of the system, we'd want to stick with system-provided pybind11 (for any deployment; CI might be a different topic).

> As to having NumPy 2 compatibility, generally if packages are built against NumPy 2, they will support NumPy 1 & 2. 

Nice, so, if we have a CI worker with a bespoke version of pybind11 and NumPy, we could test that it works, before NumPy 2 hits "mass market".

ryanvolz, 2024-06-03:
It doesn't look like it will be too much work to support or test right now with the conda-forge rollout, even if distributions lag in having NumPy 2.0 and pybind11 2.12 available.

I implemented the one-line fix I indicated above, and now I have passing builds on the conda-forge PR: https://github.com/conda-forge/gnuradio-feedstock/pull/271/checks?check_run_id=25748381780. There are still Linux failures, but those look like they have nothing to do with NumPy 2.0 and everything to do with something that changed very recently with the alsa-lib package. Once I can track that down, I can probably confirm NumPy 2.0 compatibility.

PR #7385 (2024-06-06): filter: python: Change float_ to float64 for numpy 2.0 compatibility

## Description

Numpy 2.0 is being released soon with some breaking changes. Conda-forge is already working on building against the release candidate to make as many packages available as possible when the release happens, and that has [included GNU Radio](https://github.com/conda-forge/gnuradio-feedstock/pull/271). Just this small change was necessary for the build to pass successfully.

## Related Issue

Fixes #7378

## Which blocks/areas does this affect?
closed 2024-06-06 by willcode