gnuradio/gnuradio #7384: Linux Mint MATE - Dark Mode - White Text on White Background

billwestrup, 2024-06-05:
### What happened?

Linux Mint 21.3 MATE
The ID fields in blocks are unreadable due to white text on a white background. Issue occurs with all the default MATE dark themes.

[image]

### System Information

OS: Linux Mint 21.3 MATE
GR Installation Method: installed from the system package manager

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.1.1

### Steps to Reproduce the Problem

Open GNU Radio Companion. 
Double click on the "Options" module. 
From the MATE menu chose Preferences -> Appearance. 
Select one of the dark themes. 
ID field on the options module will switch to white text on white background.

willcode, 2024-06-06:
@billwestrup There have been a number of changes in this area since 3.10.1.1, which is more than 2 years old. Have you tried a more recent version?

marcusmueller, 2024-06-06:
Solved in #6888 , changed after; closing as duplicate of #7029 #5632 #2687. Please update to a later version of GNU Radio; it's a bit surprising that the newest mint ships such an old version.

Building your own gnuradio package *should* be quite straightforward; however, I've tried, and mint packaging has diverged from both Ubuntu and debian packaging (which have recent versions). So, I wish I'd have a quick and easy solution for you, but it hinges on Mint to update their packages. If you file a UI bug with them (because, clearly, yours is a UI bug), kindly link them to this issue and tell them that upstream said they need to do:

- import debian trixie's or Ubuntu 24.04's gnuradio-3.10.10.0-~.dsc
- rename their libvolk2 to libvolk (or add aliases)
- package the (very easy to package) cppzmq-dev (should be trivially importable from debian)
- update their dpkg-dev or reduce the requirement in the gnuradio~.dsc

**Of course,** we care about *you* being able to use GNU Radio. So, if you need to get a recent version of GNU Radio on Linux mint, you can go our conda route (install miniconda, make new conda environment, activate environment, install `gnuradio` in that), as [documented in our wiki](https://wiki.gnuradio.org/index.php?title=CondaInstall#Step_1:_Install_conda_itself).

closed 2024-06-06 by marcusmueller