gnuradio/gnuradio #7390: Inclusion of apache thrift as a dependency on mingw64 causes broken libgnuradio-runtime.dll

chrisjohgorman, 2024-06-13:
### What happened?

During an attempt to install and run gnuradio on msys2/mingw64, one of the linker commands failed trying to link libgnuradio-runtime.dll.  The problem is that boost requires winsock2 under windows when using apache thrift rpc.  I attempted to go around this with the following patch.

```
diff --git a/gnuradio-runtime/lib/CMakeLists.txt b/gnuradio-runtime/lib/CMakeLists.txt
index 577262df2..553731167 100644
--- a/gnuradio-runtime/lib/CMakeLists.txt
+++ b/gnuradio-runtime/lib/CMakeLists.txt
@@ -129,7 +129,8 @@ target_link_libraries(
         MPLib::mplib
     PRIVATE
         Boost::program_options
-        libunwind::libunwind)
+        libunwind::libunwind
+        ws2_32)
 
 # Address linker issues with std::filesystem on Centos 8 and Debian
 target_link_libraries(
```
This allows for a build to proceed, but causes errors when anything tries to load gnuradio-runtime.  This meant that I could only load pmt and a couple other modules from python with a build built with this patch.  Once it was determined that thrift was the problem, I added -DENABLE_CTRLPORT_THRIFT=OFF to the command line and built without the patch and gnuradio-companion.py now loads and is usable. 

### System Information

OS: Windows 11 msys2/mingw64 latest
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-779-g5c1edfe0

### Steps to Reproduce the Problem

- Install a msys2/mingw64 x86_64 environment.
- add build-tools needed via pacman, `pacman -S --needed base-devel mingw-w64-x86_64-toolchain`
- clone my github repository `git clone https://github.com/chrisjohgorman/MINGW-packages.git` and checkout the new-packages branch `git checkout -b new-packages` for all the gnuradio dependencies.
- open a mingw64 shell and change directory to one of the gnuradio dependencies.
- build that dependency with `makepkg-mingw` and repeat until all gnuradio dependencies are built.  Then install each package with `pacman -U <package_name>.zst`
- Or optionally wait until my packages are approved for inclusion to msys2/mingw64 then install them with `pacman -S mingw-w64-x86_64-gnuradio`

The log output is a gdb backtrace of what happens when thrift and winsock2 are enabled.  The python command is `python -c "from gnuradio import gr;"`

### Relevant log output

```shell
GNU gdb (GDB) 14.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Attaching to process 6076
[New Thread 6076.0x1b48]
[New Thread 6076.0x17a4]
Reading symbols from C:\msys64\mingw64\bin\python.exe...
(No debugging symbols found in C:\msys64\mingw64\bin\python.exe)
(gdb) threads
[... 74 lines cut ...]
#64 0x00007ff6f9941406 in ?? ()
#65 0x00007fff06cc257d in KERNEL32!BaseThreadInitThunk () from C:\WINDOWS\System32\kernel32.dll
#66 0x00007fff0784aa48 in ntdll!RtlUserThreadStart () from C:\WINDOWS\SYSTEM32\ntdll.dll
#67 0x0000000000000000 in ?? ()
(gdb) thread 2
[Switching to thread 2 (Thread 6076.0x1b48)]
#0  0x00007fff0788f9d4 in ntdll!ZwWaitForSingleObject () from C:\WINDOWS\SYSTEM32\ntdll.dll
(gdb) bt
#0  0x00007fff0788f9d4 in ntdll!ZwWaitForSingleObject () from C:\WINDOWS\SYSTEM32\ntdll.dll
#1  0x00007fff0781a00e in ntdll!LdrGetDllHandleEx () from C:\WINDOWS\SYSTEM32\ntdll.dll
#2  0x00007fff078183a3 in ntdll!RtlLeaveCriticalSection () from C:\WINDOWS\SYSTEM32\ntdll.dll
#3  0x00007fff07863f6f in ntdll!LdrInitializeThunk () from C:\WINDOWS\SYSTEM32\ntdll.dll
#4  0x00007fff07863ea3 in ntdll!LdrInitializeThunk () from C:\WINDOWS\SYSTEM32\ntdll.dll
#5  0x00007fff07863dce in ntdll!LdrInitializeThunk () from C:\WINDOWS\SYSTEM32\ntdll.dll
#6  0x0000000000000000 in ?? ()
(gdb) thread 3
[Switching to thread 3 (Thread 6076.0x17a4)]
#0  0x00007fff07893641 in ntdll!DbgBreakPoint () from C:\WINDOWS\SYSTEM32\ntdll.dll
(gdb) bt
#0  0x00007fff07893641 in ntdll!DbgBreakPoint () from C:\WINDOWS\SYSTEM32\ntdll.dll
#1  0x00007fff078c6f6e in ntdll!DbgUiRemoteBreakin () from C:\WINDOWS\SYSTEM32\ntdll.dll
#2  0x00007fff06cc257d in KERNEL32!BaseThreadInitThunk () from C:\WINDOWS\System32\kernel32.dll
#3  0x00007fff0784aa48 in ntdll!RtlUserThreadStart () from C:\WINDOWS\SYSTEM32\ntdll.dll
#4  0x0000000000000000 in ?? ()
```

chrisjohgorman, 2024-06-14:
Way back in my bug report I said the branch to use was `new-packages`.  I'm sorry that was wrong.  (Late night bug writing.)  The branch is `new-pkgbuilds`.  Sorry for the error.

ryanvolz, 2024-07-08:
FYI, [thrift support doesn't build with MSVC either](https://github.com/gnuradio/gnuradio/pull/6271), so this is probably more of a general Windows problem than anything specific to the compiler. Fixing that is above my head, but I would happily glom on to a fix to enable thrift support in the Windows conda builds.

chrisjohgorman, 2024-07-11:
Yeah, it took me a while to see that I needed to exclude thrift from the build.  I too doubt that I will be able to fix this.  I was hoping that by reporting this, someone with more windows skills than I will be able to debug the problem.  It's good to know that I have some company in my lack of thrift support.  Thanks for the information @ryanvolz.