gnuradio/gnuradio #7426: Issue with iio attribute updater

paolocos, 2024-07-08:
### What happened?

simple flowgraph with QT range (for frequency), iio_attribute_updater, iio_attribute_sink.
At start seems to work but as soon as I modify the slider to change the frequency the following error occurs and gnu radio crash.
=============================================================
Executing: C:\Users\emb-user\radioconda\python.exe -u D:\Documents\grc_projects\ad9361.py

Traceback (most recent call last):
  File "C:\Users\emb-user\radioconda\Lib\site-packages\gnuradio\qtgui\range.py", line 161, in changed
    self.notifyChanged(self.rangeType(val))
  File "D:\Documents\grc_projects\ad9361.py", line 182, in set_freq
    self.set_frq(str(self.freq))
  File "D:\Documents\grc_projects\ad9361.py", line 205, in set_frq
    self.iio_attr_updater_0.set_value(self.frq)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'gnuradio.iio.iio_python.attr_updater' object has no attribute 'set_value'

>>> Done (return code 3221226505)

### System Information

OS: Windows
GR Conda
version 3.10.10.0 (Python 3.11.9)

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

QT range (for frequency), iio_attribute_updater, iio_attribute_sink.
Start it.
Move the slider.

### Relevant log output

```shell
Executing: C:\Users\emb-user\radioconda\python.exe -u D:\Documents\grc_projects\ad9361.py

Traceback (most recent call last):
  File "C:\Users\emb-user\radioconda\Lib\site-packages\gnuradio\qtgui\range.py", line 161, in changed
    self.notifyChanged(self.rangeType(val))
  File "D:\Documents\grc_projects\ad9361.py", line 182, in set_freq
    self.set_frq(str(self.freq))
  File "D:\Documents\grc_projects\ad9361.py", line 205, in set_frq
    self.iio_attr_updater_0.set_value(self.frq)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'gnuradio.iio.iio_python.attr_updater' object has no attribute 'set_value'

>>> Done (return code 3221226505)
```

bezborodow, 2024-08-02:
Myself and another user seem to be having this issue also as per: https://ez.analog.com/linux-software-drivers/f/q-a/558818/adrv9009-rx-to-tx

bezborodow, 2024-08-02:
This appears to have been caused by an issue with the Python bindings in GNU Radio, which appears to have been fixed by https://github.com/gnuradio/gnuradio/commit/70fa7e352d54a4d1afb1dde987d373e391902eaf under https://github.com/gnuradio/gnuradio/pull/7309. The relevant file is `gr-iio/python/iio/bindings/attr_updater_python.cc` where the binding `set_params` should be `set_value`.

This issue is therefore a duplicate of #7309.