gnuradio/gnuradio #7444: OOT template: move python module directory

AsriFox, 2024-07-24:
### Feature Description

In the OOT template, move `python/modname` directory to `python/gnuradio/modname`. This will make the module structure similar to the installation structure, thus simplifying python imports.

### Feature Urgency

low (just an idea)

### More Information

The module's Python libraries are installed to `$GR_PREFIX/lib/python3/dist-packages/gnuradio/modname`. If `PYTHONPATH` contains `$GR_PREFIX/lib/python3/dist-packages`, then the module can be imported as `from gnuradio import modname`.

The module project structure, however, is different: `$PROJECT_ROOT/python/modname`, so the local imports (e.g. in unit testing) are `from python import modname`.

Moving the directory will unify these two structures, allowing for simpler imports in unit testing (`from gnuradio import modname` instead of try/except blocks) and local running of examples without installing. Consider this launch command:
```sh
env LD_LIBRARY_PATH=./build/lib:./build/python/bindings:$LD_LIBRARY_PATH \
    PYTHONPATH=./build/python:$PYTHONPATH \
    GRC_BLOCKS_PATH=./grc:$GRC_BLOCKS_PATH \
  grcc -o ./build/examples -r ./examples/example.grc
```
Or this launch configuration for source debugging in VSCodium: [launch.json](https://github.com/user-attachments/files/16361928/launch.json)

marcusmueller, 2024-07-24:
I kind of like the idea to make things more consistent! Note, will need small changes to gr_modtool / blocktool (so that the parsers can find the files). That logic is due for an overhaul anyways, it tries to describe where files lie in terms of specifying the GNU Radio version, so some things are "GR38" because 3.8 moved files there, even if the version of GNU Radio is 3.10, for other things it makes a difference whether you're pre-3.9 or after, and so on, and I frankly think we don't *need* that cruft. If you're using a new gr_modtool with an old OOT, I'd think mostly to get that to be portable to a newer version of GR.

AsriFox, 2024-07-24:
I'm slightly bothered by `../../..` in [bindings/CMakeLists.txt](https://github.com/gnuradio/gnuradio/blob/2a8b58ca18b5bbd978ca6cd495450c2a9b4ea2b8/gr-utils/modtool/templates/gr-newmod/python/howto/bindings/CMakeLists.txt#L29), which will need to be changed to `../../../..`. Is this some old code, from before `${PROJECT_SOURCE_DIR}` existed?

marcusmueller, 2024-07-24:
it might be old code, but it should really be `${PROJECT_SOURCE_DIR}`