gnuradio/gnuradio #7456: `wavfile_sink` does not continue after lock/unlock

baskiton, 2024-07-30:
### What happened?

A bug happened!

### System Information

OS: Debian 12
GR Installation Method: Conda (conda-forge)

It's bug or feature? WAV file sinking stopped after lock()/unlock(). So any flowgraph reconfigure are not posible

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

```python
import pathlib
import time

import gnuradio as gr
import gnuradio.analog
import gnuradio.blocks
import gnuradio.gr

def foo(var, sr, fp):
    print(f'test {var!r}')
    tb = gr.gr.top_block('Test')
    src = gr.analog.noise_source_c(gr.analog.GR_GAUSSIAN, 1, 0)
    thr = gr.blocks.throttle(gr.gr.sizeof_gr_complex, sr)

    if var == 'file':
        sink = gr.blocks.file_sink(gr.gr.sizeof_gr_complex, str(fp))
        tb.connect(src, thr, sink)
    else:
        ctf = gr.blocks.complex_to_float()
        sink = gr.blocks.wavfile_sink(
            str(fp),
            2,
            sr,
            gr.blocks.FORMAT_WAV,
            gr.blocks.FORMAT_FLOAT,
            False
        )
        tb.connect(src, thr, ctf)
        for i in range(2):
            tb.connect((ctf, i), (sink, i))

    tb.start()

    time.sleep(3)
    tb.lock()
    st = fp.stat()
    print(f'mt={st.st_mtime:.3f} sz={st.st_size}')
    tb.unlock()

    time.sleep(3)
    tb.stop()
    tb.wait()
    st = fp.stat()
    print(f'mt={st.st_mtime:.3f} sz={st.st_size}')

if __name__ == '__main__':
    print('GNURadio', gr.gr.version())
    sr = 48000
    fp = pathlib.Path('/tmp/sink_test')
    for var in ('file', 'wav'):
        foo(var, sr, fp)

```

### Relevant log output

```shell
GNURadio 3.10.10.0
test 'file'
mt=1722329829.336 sz=1146744
mt=1722329832.408 sz=2293488
test 'wav'
mt=1722329835.492 sz=1146832
mt=1722329835.504 sz=1146832
```

baskiton, 2024-07-30:
In small projects this will work, but in larger projects, where wav_sink can be hidden deep in hier-blocks, this will become a huge crutch, which, moreover, is extremely difficult to manage

willcode, 2024-07-30:
Marked as a bug - hope the workaround helps until someone take a look at the problem.

sladecek, 2024-11-16:
I am working on it.

sladecek, 2024-11-24:
I tried to approach this issue as an opportunity to learn something about GNU Radio.  

The flowgraph uses the `stop()` method of a block to indicate that it has been locked for configuration. I believe the correct behavior for file-based sinks is to write all data and headers to disk, flush the file buffers, but avoid closing the file.  

This behavior is implemented in the `file_sink`. Since the `file_sink` is very simple and does not require updating headers, flushing the file is sufficient.  

On the other hand, `wavfile_sink` can write to several file formats, some of which include headers. The implementation relies on [libsndfile](https://libsndfile.github.io/libsndfile/), which provides a [flush function](https://libsndfile.github.io/libsndfile/api.html#write_sync). However, the source code indicates that this function does not update headers before flushing.  

The current implementation of `wavfile_sink` calls the `close()` method within the `stop()` method to ensure the file on disk is correct. However, this also closes the file.  

My solution is to add a flag for automatic reopening. When the file is opened, its filename is stored. When the `stop()` method is called, the file is closed, and the automatic reopen flag is set. Later, if a write operation is initiated (via `work()`), and the flag is set, the file is automatically reopened in append mode.  

I hope this resolves the issue.

The PR will be submitted after PR #7550 is resolved.

willcode, 2024-11-24:
If we want to do this, using `start()`, and moving the necessary logic from the constructor to `start()` would be better.

Also, it's questionable whether a sound file needs to be readable when `stop()` is called. `close()` could just be called in the destructor.

sladecek, 2024-11-24:
OK, I can place the reopening code within `start()`. Let’s postpone this part of the discussion until the code review.  

I believe all file-based sinks should have consistent behavior, and the `file_sink` calls flush. This was discussed in [[this issue](https://github.com/gnuradio/gnuradio/issues/2590)](https://github.com/gnuradio/gnuradio/issues/2590).

sladecek, 2024-12-03:
Could someone please review my PR?

marcusmueller, 2024-12-05:
Hi @sladecek, sorry that we haven't found time yet to do that. Super busy times; please don't assume we're not thankful (we are, very much!) for the contribution.

PR #7556 (2024-12-29): blocks: Reopen wavfile_sink after stop().
## Description

The flowgraph uses the stop() method of a block 
to indicate that it has been locked for
configuration. I believe the correct behavior for
file-based sinks is to write all data and headers
to disk, flush the file buffers, but avoid closing the file 
(see also #2590).

This behavior is implemented in the
file_sink. Since the file_sink is very simple and
does not require updating headers, flushing the
file is sufficient.

On the other hand, wavfile_sink can write to
several file formats, some of which include
headers. The implementation r [... cut]
commit 64d2d123 (2024-12-29): blocks: Reopen wavfile_sink after stop().

Fixes #7456 .

The flowgraph uses the stop() method of a block to
indicate that it has been locked for
configuration. I believe the correct behavior for
file-based sinks is to write all data and headers
to disk, flush the file buffers, but avoid closing
the file (see also #2590).

This behavior is implemented in the
file_sink. Since the file_sink is very simple and
does not require updating headers, flushing the
file is sufficient.

On the other hand, wavfile_sink can write to
several file formats, some of which include
headers. The implementation rel [... cut]
commit 35346914 (2025-01-14): blocks: Reopen wavfile_sink after stop().

Fixes #7456 .

The flowgraph uses the stop() method of a block to
indicate that it has been locked for
configuration. I believe the correct behavior for
file-based sinks is to write all data and headers
to disk, flush the file buffers, but avoid closing
the file (see also #2590).

This behavior is implemented in the
file_sink. Since the file_sink is very simple and
does not require updating headers, flushing the
file is sufficient.

On the other hand, wavfile_sink can write to
several file formats, some of which include
headers. The implementation rel [... cut]
closed 2024-12-29 by willcode