gnuradio/gnuradio #7460: maint-3.10: fatal error: burst_to_stream_pydoc.h: No such file or directory

wavewalkerdsp, 2024-08-03:
### What happened?

Compilation on latest on maint-3.10 fails (git rev ee27d6f3d63b39e1427f80de9950d9a64bdae096)

lto-wrapper: warning: using serial compilation of 5 LTRANS jobs
/home/user/gnuradio-install/gnuradio/gr-blocks/python/blocks/bindings/burst_to_stream_python.cc:28:10: fatal error: burst_to_stream_pydoc.h: No such file or directory
   28 | #include <burst_to_stream_pydoc.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [gr-blocks/python/blocks/bindings/CMakeFiles/blocks_python.dir/build.make:349: gr-blocks/python/blocks/bindings/CMakeFiles/blocks_python.dir/burst_to_stream_python.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:5380: gr-blocks/python/blocks/bindings/CMakeFiles/blocks_python.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

Likely related to commit a5078c8e19f5184461d0c018cd596777c19e2542 which appears to have created the new burst_to_stream block.

### System Information

OS: Ubuntu 22.04
GR Installation Method: source-based installation via github

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

$ git checkout maint-3.10
$ git pull
$ cd build/
$ cmake -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3 ../
$ make -j4

willcode, 2024-08-03:
Is this with a clean build, i.e., no `./build` subdir?

wavewalkerdsp, 2024-08-03:
Good catch, works fine when I run out of a new build directory. Closing.

closed 2024-08-03 by wavewalkerdsp