gnuradio/gnuradio #7487: constellation_soft_decoder.grc has mismatched 16-QAM modulation

wavewalkerdsp, 2024-08-28:
### What happened?

Reviewing in-tree flowgraph constellation_soft_decoder.grc and all of the modulation pairs work correctly with exception of 16-QAM. The flowgraph allows BPSK, QPSK and 16-QAM modulations to be selected, on both bottom left and bottom right, to ensure they are the same. BPSK and QPSK work correctly, but 16-QAM does not.

[image]

[image]

### System Information

OS: (Linux Distro, Mac, Windows)
GR Installation Method: (native distro packages - apt, rpm, etc., Conda, Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

v3.11.0.0git-809-g5c30d92b

### Steps to Reproduce the Problem

Open gr-digital/examples/demod/constellation_soft_decoder.grc, enable both 16-QAM variable blocks and disable all BPSK and QPSK variable blocks, run flowgraph.

ffu, 2025-11-20:
While working on this fix, I tested with other constellations in psk_constellations.py and qam_constellations.py. There are several that fail to work with the aforementioned grc example. It turns out there is a misunderstanding of a constellation's symbol mapping or encoding, which may not be related to its decoding.

For many constellations, this is not a problem because the map for the encoding and decoding are the same. For those that are not (e.g. psk_4_5 or psk_4_6) the soft decoder does not work. Additional fixes are needed to make the soft decoder to work. Additional documentation could also make clear how the symbol map should be used.

ffu, 2026-10-13:
There are a lot of problems with soft decoding and the external lookup table. I wouldn't be opposed to deprecation since maintenance may require too much effort. For starters, we could delete the constellation object that use external lookup tables in the example flowgraph. Issue would be resolved with much less software changes.