gnuradio/gnuradio #7493: gr-vocoder/examples/grfreedv.grc missing blocks

wavewalkerdsp, 2024-08-29:
### What happened?

The in-tree flowgraph gr-vocoder/examples/grfreedv.grc references blocks vocoder_freedv_tx_ss, vocoder_freedv_rx_ss which do not appear to exist anymore

[image]

### System Information

OS: Ubuntu 24.04
GR Installation Method: source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-809-g5c30d92b

### Steps to Reproduce the Problem

Viewable in flowgraph

drmpeg, 2024-08-29:
You're probably just missing the dependencies. If you look at the CMake output, you'll see:
```
-- Configuring gr-vocoder support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_FFT = ON
--   Dependency ENABLE_GR_BLOCKS = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency ENABLE_GR_ANALOG = ON
--   Enabling gr-vocoder support.
--   Override with -DENABLE_GR_VOCODER=ON/OFF
-- Could NOT find Codec2 (missing: LIBCODEC2_LIBRARIES LIBCODEC2_INCLUDE_DIRS) 
-- Could NOT find GSM (missing: LIBGSM_LIBRARIES LIBGSM_INCLUDE_DIRS) 
```
But gr-vocoder still gets enabled. So try:

```sudo apt-get install libcodec2-dev```

wavewalkerdsp, 2024-08-30:
@drmpeg you're correct, was missing that dependency, thank you. issue resolved, closing.

closed 2024-08-30 by wavewalkerdsp