gnuradio/gnuradio #7494: gr-vocoder/examples/loopback-codec2.grc flowgraph missing blocks

wavewalkerdsp, 2024-08-29:
### What happened?

In-tree flowgraph gr-vocoder/examples/loopback-codec2.grc is missing blocks vocoder_codec2_encode_sp, vocoder_codec2_decode_sp, blocks do not seem to exist.

[image]

### System Information

OS: (Linux Distro, Mac, Windows)
GR Installation Method: (native distro packages - apt, rpm, etc., Conda, Source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-809-g5c30d92b

### Steps to Reproduce the Problem

Viewable in flowgraph

drmpeg, 2024-08-30:
Should also be resolved with installing libcodec2-dev.

wavewalkerdsp, 2024-08-31:
@drmpeg yes thank you, also closing this one

closed 2024-08-31 by wavewalkerdsp