gnuradio/gnuradio #7495: uhd_msg_tune.grc does not update frequency axis

wavewalkerdsp, 2024-08-29:
### What happened?

The flowgraph gr-uhd/examples/grc/uhd_msg_tune.grc does not update the frequency axis in QT GUI Freq Sink when UHD is retuned. The QT GUI Freq Sink does not have message input enabled, and when it does and connected to retune message, it still does not update.

[image]

### System Information

OS: Ubuntu 24.04
GR Installation Method: source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-809-g5c30d92b

### Steps to Reproduce the Problem

Run flowgraph, look at the frequency axis when retuning.

CanoRoger, 2024-10-24:
Most likely related, since the addition of the async_message output in the "UHD: USRP Source block", it is not taking anymore input commands. 

Can't tune frequency (and other requests) on USRP using PMTs. 

This worked in 3.10.1.1 and previous versions.
Not anymore at least tested from 3.10.11.0

On X300, UHD-3.15 and 4.1.

marcusmueller, 2024-12-14:
@CanoRoger  interesting! Could you open a new issue "USRP source doesn't add tags when tuning via cmd port" or such? I think we need to track this separately :)

closed 2025-01-30 by willcode