gnuradio/gnuradio #7504: segfault on startup while reading previously opened grc containing OOT python QT sink

jabowery, 2024-09-05:
### What happened?

Attempted to bring up `gnuradio-companion` but it crashed with a `Segmentation fault (core dumped)`.

### System Information

OS: Ubuntu 22.04.4 LTS
GR Installation Method: apt

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.1.1

### Steps to Reproduce the Problem

1. Download this file: [minimalexample.grc.txt](https://github.com/user-attachments/files/16896595/minimalexample.grc.txt)
2. Rename it to minimalexample.grc
3. Bring up `gnuradio-companion`
4. File -> Open -> minimalexample.grc
5. Run -> Execute 
6. Exit gnuradio-companion`
7. Bring up `gnuradio-companion`

### Relevant log output

```shell
gnuradio-companion 
>>> Warning: vocoder_codec2_decode_ps - option_attributes are for enums only, ignoring
>>> Warning: vocoder_codec2_encode_sp - option_attributes are for enums only, ignoring
<<< Welcome to GNU Radio Companion 3.10.1.1 >>>

Block paths:
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/jabowery/devel/weber/steffen/simulations/minimalexample.grc"
>>> Done
Segmentation fault (core dumped)
```

dl1ksv, 2024-09-06:
I tested your example with a fresh build from main with fedora. After adding the pyqtgraph package I was able to build and run your example. Both versions grc --gtk and grc --qt worked with some error messages.

grc --gtk
<<< Welcome to GNU Radio Companion v3.11.0.0git-813-g7d5e5f97 >>>

Block paths:
	/home/schroer/.local/state/gnuradio
	/usr/local/oot/share/gnuradio/grc/blocks
	/usr/local/gnuradio/share/gnuradio/grc/blocks

Loading: "/home/schroer/gnuradiocomponents/Test/basic_for_hier.grc"
>>> Done

(gnuradio-companion:3472): Gtk-WARNING **: 12:02:29.641: Failed to measure available space: Der angegebene Ort wird nicht unterstützt

(gnuradio-companion:3472): Gtk-WARNING **: 12:02:29.642: Failed to measure available space: Der angegebene Ort wird nicht unterstützt

Loading: "/home/schroer/gnuradiocomponents/bugs/minimalexample.grc"
>>> Done

Generating: "/home/schroer/gnuradiocomponents/bugs/SimulateFM103p1DownConverted.py"
>>> Warning: The block 'blocks_throttle_0' is deprecated.

Generating: "/home/schroer/gnuradiocomponents/bugs/SimulateFM103p1DownConverted.py"
>>> Warning: The block 'blocks_throttle_0' is deprecated.

Executing: /usr/bin/python3 -u /home/schroer/gnuradiocomponents/bugs/SimulateFM103p1DownConverted.py

log :warning: Matplotlib is a required dependency to use DistanceRadar and AzElPlot.  Please install matplotlib to use these blocks (https://matplotlib.org/)
vmcircbuf_sysconfig :info: Using gr::vmcircbuf_sysv_shm_factory
QObject::killTimer: Timers cannot be stopped from another thread
QObject::setParent: Cannot set parent, new parent is in a different thread
QObject::installEventFilter(): Cannot filter events for objects in a different thread.
QCoreApplication::sendPostedEvents: Cannot send posted events for objects in another thread

>>> Done

willcode, 2024-09-09:
Embedded python block in this example creates windows, which is a bit outside the intended use. Not that there's any explicit limit to what you can do with embedded blocks, but you're better off creating an OOT Python block once display, threads/timer, and more complex things are involved.

We're not going to be able to figure it out from here, so closing. What is likely happing is that the last active flowgraph (this one?)  is parsed when grc starts. Embedded Python blocks are immediately evaluated, and something is going wrong at that point.

closed 2024-09-09 by willcode