gnuradio/gnuradio #7528: block::set_processor_affinity() will crash after the block was stopped.

EliteWeapon, 2024-10-18:
### What happened?

Calling **set_processor_affnity()** on a block which was stopped will cause progam to crash.

### System Information

OS: Linux Distro
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.11.0

### Steps to Reproduce the Problem

1. make a top block with one or more sub-blocks, then start it.
2. stop the top block for modifying some sub-blocks.
3. call set_processor_affinity on the top block. this step will cause the program to crash.

marcusmueller, 2024-10-18:
Can't reproduce on main with 
```python
#!/usr/bin/env python3
from gnuradio import gr
from gnuradio import blocks
from time import sleep

tb = gr.top_block()

src = blocks.null_source(1)
sink = blocks.null_sink(1)

tb.connect(src, sink)

tb.start()

sleep(0.5)

tb.stop()

tb.set_processor_affinity([0, ])
```

so building 3.10.11.0 from source on Fedora 40 x86_64. 

@EliteWeapon any specific info on how it crashes, maybe even a debugger backtrace?

marcusmueller, 2024-10-18:
Can't reproduce on v3.10.11.0 either.  @EliteWeapon that puts the onus of giving us something minimal that can reproduce the issue, and providing us with more information on your platform, on you, I'm afraid.

EliteWeapon, 2024-10-18:
I checked the source codes:
1. When starting a top block, its sub-block detail's memeber '**threaded**' and '**thread**' will be set to valid values in funtion **tpb_thread_body()**.
```
tpb_thread_body::tpb_thread_body(block_sptr block,
                                 gr::thread::barrier_sptr start_sync,
                                 int max_noutput_items)
    : d_exec(block, max_noutput_items)
{
#if defined(_MSC_VER) || defined(__MINGW32__)
#include <windows.h>
    thread::set_thread_name(GetCurrentThread(),
                            block->name() + std::to_string(block->unique_id()));
#else
    thread::set_thread_name(pthread_self(),
                            block->name() + std::to_string(block->unique_id()));
#endif

    block_detail* d = block->detail().get();
    block_executor::state s;
    pmt::pmt_t msg;

    d->threaded = true;
    d->thread = gr::thread::get_current_thread_id();

    ...

    start_sync->wait();
    while (1) {
        boost::this_thread::interruption_point();
        ...
     }
}

```
2. Then, we stop the top block. As the result,  the sub-block's underlying thread will be destroyed.
```
void top_block_impl::stop()
{
    ...
    if (d_scheduler)
        d_scheduler->stop();
    ...
}
```
`
void scheduler_tpb::stop() { d_threads.interrupt_all(); }
`
3. But the sub-block detail's member '**threaded**' and '**thread**' were left unchanged and now become invalid.
4. Here, we continue to call **block::set_processor_affinity()** on the sub-block and this will cause to further call **block_detail::set_processor_affinity()**:
```
void block_detail::set_processor_affinity(const std::vector<int>& mask)
{
    if (threaded) {
        try {
            gr::thread::thread_bind_to_processor(thread, mask);
        } catch (std::runtime_error& e) {
            d_logger->error("set_processor_affinity: invalid mask.");
        }
    }
}
```
Because the variable 'thread' is invalid, '**gr::thread::thread_bind_to_processor()**' will cause program to crash.

I found a solution to fix this bug: 
1) add a RAII guard for block detail's thread information
2) the guard's destructor will be called to assure the thread information correct when the block's underlying thread exit.
```
// guard for thread information of block.detail
class block_thread_info_guard
{
public:
    block_thread_info_guard(const block_sptr& block)
    {
        d_block = block;

        if (d_block) {
            block_detail* d = d_block->detail().get();
            if (d) {
                d->threaded = true;
                d->thread = gr::thread::get_current_thread_id();
            }
        }
    }

    ~block_thread_info_guard()
    {
        if (d_block) {
            block_detail* d = d_block->detail().get();
            if (d) {
                d->threaded = false;
                d->thread = gr::thread::INVALID_THREAD_ID;
            }
        }
    }

protected:
    block_sptr d_block;
};

```
```
tpb_thread_body::tpb_thread_body(block_sptr block,
                                 gr::thread::barrier_sptr start_sync,
                                 int max_noutput_items)
    : d_exec(block, max_noutput_items)
{
#if defined(_MSC_VER) || defined(__MINGW32__)
#include <windows.h>
    thread::set_thread_name(GetCurrentThread(),
                            block->name() + std::to_string(block->unique_id()));
#else
    thread::set_thread_name(pthread_self(),
                            block->name() + std::to_string(block->unique_id()));
#endif

    block_detail* d = block->detail().get();
    block_executor::state s;
    pmt::pmt_t msg;

    // Use RAII to set and clear block's thread information
    block_thread_info_guard guard(block);

    ...

    start_sync->wait();
    while (1) {
        boost::this_thread::interruption_point();
        ...
    }
}

```

marcusmueller, 2024-10-18:
Thanks for the extensive look and proposed solution. Interestingly I still can't make the thing crash, but your reasoning does seem valid.

marcusmueller, 2024-10-18:
marking as medium (because impact is limited, but still seems to be serious enough)

marcusmueller, 2024-10-18:
really wonder why this is not making my machine crash but yours; my guess is that on my machine, the thread lingers on? Or I have a different libc? Can you please give me some insight into specifically which platform you're on?

EliteWeapon, 2024-10-23:
@marcusmueller

EliteWeapon, 2024-10-23:
I write a test program in c++ as the following:

```
#include <gnuradio/types.h>
#include <gnuradio/top_block.h>
#include <gnuradio/blocks/null_source.h>
#include <gnuradio/blocks/null_sink.h>
#include <thread>
#include <chrono>

int main() {

    auto tp_blk = gr::make_top_block("my_top_block");

    auto src_blk = gr::blocks::null_source::make(1);
    auto snk_blk = gr::blocks::null_sink::make(1);

    tp_blk->connect(src_blk, 0, snk_blk, 0);

    tp_blk->start();

    std::this_thread::sleep_for(std::chrono::seconds(1));

    tp_blk->stop();
    tp_blk->wait();

    tp_blk->set_processor_affinity(gr_vector_int{0, 1});

    return 0;
}

```

The output is:
```
block_detail :error: set_processor_affinity: invalid mask.
block_detail :error: set_processor_affinity: invalid mask.
```
The logs above appeared after ' tp_blk->set_processor_affinity(gr_vector_int{0, 1}) ' was executed.  

Furtherly, I checked the source code where the messages were logged:

```
void block_detail::set_processor_affinity(const std::vector<int>& mask)
{
    if (threaded) {
        try {
            gr::thread::thread_bind_to_processor(thread, mask);
        } catch (std::runtime_error& e) {
            d_logger->error("set_processor_affinity: invalid mask.");
        }
    }
}

```

Because the member 'thread' of block_detail is invalid after block was stopped, an exception was throwed and captured sequentially.

In this demo, progam didn't crash but logged some errors.

EliteWeapon, 2024-10-23:
But in another test, I used gqrx-2.17.5 + gr-osmosdr-0.2.6 to simulate the error.

I modified the **receiver:stop()** function in receiver.cpp as :

```
void receiver::stop()
{
    if (d_running)
    {
        tb->stop();
        tb->wait(); // If the graph is needed to run again, wait() must be called after stop
        d_running = false;

        /********* next line was newly added  *********/
        tb->set_processor_affinity(gr_vector_int{0, 1});
    }
}

```

Then I used a filesource block to make the program work. 
[image]

When I pushed the button to start  and then stop the DSP progressing, the program crashed:
[image]

```
jwp@bh-jwp:~/project_work/gqrx-2.17.5/build/src$ gdb ./gqrx
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./gqrx...
(gdb) r
Starting program: /home/jwp/project_work/gqrx-2.17.5/build/src/gqrx 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[New Thread 0x7ffff0fc4640 (LWP 88113)]
[New Thread 0x7fffeb49c640 (LWP 88114)]
[New Thread 0x7fffeac7c640 (LWP 88115)]
[New Thread 0x7fffea442640 (LWP 88116)]
[New Thread 0x7fffe9c41640 (LWP 88117)]
[New Thread 0x7fffe8faf640 (LWP 88118)]
[New Thread 0x7fffcffff640 (LWP 88120)]
[Thread 0x7fffcffff640 (LWP 88120) exited]
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.11.0
built-in source types: file rtl rtl_tcp rfspace airspyhf redpitaya 
rx_nb_cc :info: set_min_output_buffer on block 10 to 32768
Resampling audio 96000 -> 48000
[New Thread 0x7fffcffff640 (LWP 88121)]
BandPlanFile is /home/jwp/.config/gqrx/bandplan.csv
BookmarksFile is /home/jwp/.config/gqrx/bookmarks.csv
[New Thread 0x7fffd0bff640 (LWP 88122)]
[Thread 0x7fffd0bff640 (LWP 88122) exited]
[New Thread 0x7fffd0bff640 (LWP 88123)]
[Thread 0x7fffd0bff640 (LWP 88123) exited]
[New Thread 0x7fffd0bff640 (LWP 88124)]
[Thread 0x7fffeac7c640 (LWP 88115) exited]
[... 83 lines cut ...]
#20 0x00007ffff7374364 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff66b9e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007ffff7372e47 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007ffff73c8d40 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff73cbfd5 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff736c713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007ffff66b9e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff6b41307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#28 0x00007ffff6b16a2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#29 0x00007ffff1aebd6e in  () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#30 0x00007ffff4451d3b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007ffff44a72b8 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007ffff444f3e3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#33 0x00007ffff67130b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#34 0x00007ffff66b875b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#35 0x00007ffff66c0cf4 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#36 0x00005555556724f7 in main(int, char**) (argc=1, argv=0x7fffffffe0e8) at /home/jwp/project_work/gqrx-2.17.5/src/applications/gqrx/main.cpp:161
(gdb) 

```

I don't know why the exception was not captured in this test but was captured in above test.

However, the fault when '**set_processor_affinity()**' was called after the block was stopped exists.