gnuradio/gnuradio #7533: Macports installation broken, boost171 fails to build

Hritik14, 2024-10-28:
### What happened?

Hello, I'm unable to install ``gnuradio`` via macports on M1 mac running Sonoma. It fails to build boost171.

Here's the detailed log: https://gist.github.com/Hritik14/4c8b27def3a8ff29623bb4d3bc81f22b

### System Information

OS: Mac Sonoma
GR Installation Method: macports

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

sudo port install gnuradio

### Relevant log output

```shell
; sudo port selfupdate && sudo port upgrade outdated && sudo port install gnuradio
--->  Checking for newer releases of MacPorts
MacPorts base version 2.10.2 installed,
MacPorts base version 2.10.2 available.
--->  MacPorts base is already the latest version
--->  Updating the ports tree

The ports tree has been updated.
All installed ports are up to date.
Nothing to upgrade.
--->  Scanning binaries for linking errors
--->  No broken files found.
--->  No broken ports found.
--->  Computing dependencies for gnuradio
The following dependencies will be installed:
 OpenBLAS
 adwaita-icon-theme
 assimp
 at-spi2-atk
 at-spi2-core
 atk
 boost171
 boost176
 boost181
 cargo
 clang-17
 clang-18
 codec2
 cppzmq
 dbus
 dbus-glib
 dbus-python39
 double-conversion
 fftw-3
 fftw-3-single
 flac
 flite
 fonttools_select
 gcc14
 gcc14-libcxx
 gtk3
 gzip
 hicolor-icon-theme
 iso-codes
 lame
 libepoxy
 libgcc
 libgcc14
 libgit2
 libmng
 libogg
 libopus
 librsvg
 libsamplerate
 libsdl
 libsdl2
 libsndfile
 libsodium
 libssh2
 libvorbis
 llvm-17
 llvm-18
 lpcnetfreedv
 minizip
 mpg123
 mypy_select
 orc
 poppler
 pugixml
 py310-build
 py310-calver
 py310-certifi
 py310-charset-normalizer
 py310-cython
 py310-docutils
 py310-editables
 py310-flit_core
 py310-hatch-vcs
 py310-hatchling
 py310-idna
 py310-installer
 py310-mako
 py310-markupsafe
 py310-meson
 py310-meson-python
 py310-numpy
 py310-oldest-supported-numpy
 py310-packaging
 py310-pathspec
 py310-pluggy
 py310-pybind11
 py310-pyproject_hooks
 py310-pyproject_metadata
 py310-requests
 py310-roman
 py310-setuptools
 py310-setuptools_scm
 py310-six
 py310-tomli
 py310-trove-classifiers
 py310-urllib3
 py310-wheel
 py312-cython-compat
 py312-yaml
 py39-Pillow
 py39-alabaster
 py39-babel
 py39-beniget
 py39-brotli
 py39-build
 py39-cairo
 py39-calver
 py39-certifi
 py39-cffi
 py39-charset-normalizer
 py39-cheetah
 py39-click
 py39-click-plugins
 py39-contourpy
 py39-cppy
 py39-cycler
 py39-cython
 py39-cython-compat
 py39-dateutil
 py39-docutils
 py39-editables
 py39-flit_core
 py39-fonttools
 py39-gast
 py39-gobject3
 py39-hatch-vcs
 py39-hatchling
 py39-idna
 py39-imagesize
 py39-importlib-metadata
 py39-importlib-resources
 py39-installer
 py39-jinja2
 py39-kiwisolver
 py39-lxml
 py39-mako
 py39-markupsafe
 py39-matplotlib
 py39-meson
 py39-meson-python
 py39-mypy
 py39-mypy_extensions
 py39-numpy
 py39-oldest-supported-numpy
 py39-olefile
 py39-opengl
 py39-opengl-accelerate
 py39-packaging
 py39-parsing
 py39-pathspec
 py39-pluggy
 py39-ply
 py39-pybind11
 py39-pycparser
 py39-pygments
 py39-pyproject_hooks
 py39-pyproject_metadata
 py39-pyqt-builder
 py39-pyqt5
 py39-pyqt5-sip
 py39-pyqtgraph
 py39-pythran
 py39-requests
 py39-roman
 py39-scipy
 py39-setuptools
 py39-setuptools_scm
 py39-sip
 py39-six
 py39-snowballstemmer
 py39-sphinx
 py39-sphinxcontrib-applehelp
 py39-sphinxcontrib-devhelp
 py39-sphinxcontrib-htmlhelp
 py39-sphinxcontrib-jsmath
 py39-sphinxcontrib-qthelp
 py39-sphinxcontrib-serializinghtml
 py39-tkinter
 py39-tomli
 py39-trove-classifiers
 py39-types-psutil
 py39-types-setuptools
 py39-typing_extensions
 py39-tz
 py39-unicodedata2
 py39-urllib3
 py39-wheel
 py39-yaml
 py39-zipp
 py39-zmq
 py39-zopfli
 python39
 python3_select-39
 python_select-39
 qhull
 qt5
 qt5-qt3d
 qt5-qtbase
 qt5-qtconnectivity
 qt5-qtdeclarative
 qt5-qtgamepad
 qt5-qtgraphicaleffects
 qt5-qtimageformats
 qt5-qtlocation
 qt5-qtmacextras
 qt5-qtmultimedia
 qt5-qtnetworkauth
 qt5-qtquickcontrols
 qt5-qtquickcontrols2
 qt5-qtremoteobjects
 qt5-qtscript
 qt5-qtscxml
 qt5-qtsensors
 qt5-qtserialbus
 qt5-qtserialport
 qt5-qtspeech
 qt5-qtsvg
 qt5-qttools
 qt5-qttranslations
 qt5-qtwebchannel
 qt5-qtwebsockets
 qt5-qtxmlpatterns
 qt5-sqlite-plugin
 qwt-qt5
 rust
 sdl12-compat
 speex
 speexdsp
 tcl
 texlive-basic
 texlive-bin
 texlive-latex
 tk
 uhd
 vala
 volk
 xorg-libXScrnSaver
 xorg-libXcomposite
 xorg-libXcursor
 xorg-libXtst
 zmq
Continue? [Y/n]:
--->  Building boost171
Error: Failed to build boost171: command execution failed
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_boost171/boost171/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets if you believe there is a bug.
Error: Processing of port gnuradio failed
;
```

marcusmueller, 2024-10-28:
Sorry,  this isn't the place to report macports build failures! We don't do these. 

 I'd assume the macports project, however, would be happy about a report.

closed 2024-10-28 by marcusmueller

Hritik14, 2024-10-29:
Found a ticket here:  https://trac.macports.org/ticket/71159
doesn't seem to have a fix yet

marcusmueller, 2024-10-29:
well, it's 6 days old :)

temmink, 2024-12-10:
I'm having the same problem, would like to use version boost181 if possible. Where can I find the file so I can adjust the version/location check?

marcusmueller, 2024-12-10:
Hi! This is a bug tracker, and not a support forum, and I've pointed to the correct authority on macports builds (which we aren't, sorry!). 

Locking this issue to avoid attracting more "me, too!" replies. We're super happy to help people on our mailing list and our chat server :)