gnuradio/gnuradio #7536: File source/sink and TCP source/sink return garbled data

FTC55, 2024-11-02:
### What happened?

I was testing the demo flowgraph for QPSK modulation and demodulation found here https://wiki.gnuradio.org/images/d/d0/Qpsk_stage6.grc and correctly observed that the bits generated by ther Random Source are correctly decoded at the output as shown in the Time Sink. 

However, whenever I try to replicate the experiment with a File Source and File Sink or TCP Source and TCP sink, the recieved bits match with the transmitted bits just like in the example, but the data is completely garbled, which leads me to believe that this is an issue with the source and sink blocks.

I tried swapping endianness, correctly repacking bits where required, but to no avail.

### System Information

OS: Ubuntu 24.04 LTS
GR Installation Method: apt

### GNU Radio Version

4.0-git (dev-4.0)

### Steps to Reproduce the Problem

Substitute the Random Source block in the demo flowgraph with a File Source and add an output File Sink (after repacking the bits correctly). 

The generated output file will be completely different from the input file used.

marcusmueller, 2024-12-12:
Hi @FTC55; if you installed GNU Radio via apt, you definitely do not have GR 4.0-git (we don't have any packaging for that yet). So, could you please check which version of GNU Radio you have, exactly? You mention a GRC file, so best use the GRC's `Help`->`About` menu. Thank you!

marcusmueller, 2024-12-12:
It's pretty unlikely that the file source/sink or TCP source/sink modify data. These really just take the data and transport them to a file / a socket as given, without ever touching it. There's probably something else at work here! And that's why having specific information on the version is so important for reproduction. 

We'd really also need to know where specifically you use these blocks in your modification of the flow graph (I'm having multiple wild guesses, and me wildly trying different things isn't going to help you, I'm afraid). Could you please add a screenshot of the flowgraph with your modifications?

marcusmueller, 2025-01-18:
I suspect this was a problem on the user's side, and we have contradicting version info and no clarification for > 1 month: I'll go ahead and close this.

closed 2025-01-18 by marcusmueller