gnuradio/gnuradio #7537: Cannot find OSMOCOM sink widget.

DerrickLau, 2024-11-05:
### What happened?

I can no longer find the OSMOCOM sink widget for representing my HackRF One.  I used the Soapy HackRF Sink instead.  Does the Soapy sink allow for transmitting voice over the HackRF?

### System Information

OS: Debian release 12.  Codenamed Bookworm.
GR Installation Method: Used sudo apt install gnuradio.
GNURadio v3.10.5.1.

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.5.11

### Steps to Reproduce the Problem

Create a simple FM transmitter flowgraph: drag on audio source, sampling rate 48000, then NFM for Narrow Frequency Modulator, then Soapy HackRF One.  Searched for generic OSMOCOM sink but could not find it.

### Relevant log output

```shell
n/a
```

closed 2024-11-06 by marcusmueller

DerrickLau, 2024-11-06:
Sorry, where do I find the voice modulator element in GNU radio's right side panel of components?

marcusmueller, 2024-11-07:
This is a bug tracker, not a help forum, I'm afraid. Please refer to the mailing list or chat; see https://www.gnuradio.org/community/