gnuradio/gnuradio #7538: CMake Error Python bindings for channelizer.h are out of sync

eabase, 2024-11-06:
### What happened?

I'm trying to OOT compile the [gr-lora](https://github.com/rpp0/gr-lora), using MSYS / MINGW64.
CMake then encounters the following error:

```bash
#...
CMake Error at C:/msys64/mingw64/lib/cmake/gnuradio/GrPybind.cmake:238 (message):
  Python bindings for channelizer.h are out of sync
Call Stack (most recent call first):
  python/bindings/CMakeLists.txt:40 (GR_PYBIND_MAKE_OOT)
```

### System Information

Windows-11 Pro
MSYS / MINGW64

Installed with: 
`pacman -S mingw64/mingw-w64-x86_64-gnuradio`

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.11.0

### Steps to Reproduce the Problem

Install all `pacman` dependecies:

```bash
pacman -S mingw64/mingw-w64-x86_64-gnuradio
pacman -S mingw64/mingw-w64-x86_64-doxygen
pacman -S mingw64/mingw-w64-x86_64-graphviz
pacman -S mingw64/mingw-w64-x86_64-pybind11
pacman -S mingw64/mingw-w64-x86_64-python-pygccxml
```

clone and run

```bash
git clone https://github.com/rpp0/gr-lora.git
cd gr-lora && mkdir build && cd build
cmake ../ -Wno-dev --fresh -DCMAKE_INSTALL_PREFIX=./opt -DPYTHON_EXECUTABLE=/mingw64/bin/python.exe -DCMAKE_CURRENT_SOURCE_DIR=../
```

### Relevant log output

```shell
# OUTPUT:  -----------------------------------------------------------
$ cmake ../ -Wno-dev --fresh -DCMAKE_INSTALL_PREFIX=./opt -DPYTHON_EXECUTABLE=/mingw64/bin/python.exe -DCMAKE_CURRENT_SOURCE_DIR=../
-- Building for: Ninja
-- The CXX compiler identification is GNU 14.2.0
-- The C compiler identification is GNU 14.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/msys64/mingw64/bin/c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found PkgConfig: C:/msys64/mingw64/bin/pkg-config.exe (found version "2.3.0")
-- Checking for module 'gmp'
--   Found gmp, version 6.3.0
-- Found GMP: C:/msys64/mingw64/lib/libgmpxx.dll.a
-- Using GMP.
-- Found MPLIB: C:/msys64/mingw64/lib/libgmpxx.dll.a
-- Found Boost: C:/msys64/mingw64/lib/cmake/Boost-1.86.0/BoostConfig.cmake (found suitable version "1.86.0", minimum required is "1.85.0") found components: date_time program_options system regex thread unit_test_framework
-- Found Volk: Volk::volk
-- User set python executable D:/a/msys64/mingw64/bin/python3.exe
-- Found PythonInterp: C:/msys64/mingw64/bin/python.exe (found version "3.11.10")
-- Found PythonLibs: C:/msys64/mingw64/lib/libpython3.11.dll.a (found suitable exact version "3.11.10")
-- Found Git: C:/Program Files/Git/cmd/git.exe
-- Extracting version information from git describe...
-- Found Doxygen: C:/msys64/mingw64/bin/doxygen.exe (found version "1.12.0") found components: doxygen dot
-- No C++ unit tests... skipping
-- PYTHON and GRC components are enabled
-- Python checking for pygccxml - found
-- Found PythonInterp: C:/msys64/mingw64/bin/python.exe
-- Found PythonLibs: C:/msys64/mingw64/lib/libpython3.11.dll.a
-- pybind11::lto disabled (problems with undefined symbols for MinGW for now)
-- pybind11::thin_lto disabled (problems with undefined symbols for MinGW for now)
-- Found pybind11: C:/msys64/mingw64/include (found version "2.13.6")

CMake Error at C:/msys64/mingw64/lib/cmake/gnuradio/GrPybind.cmake:238 (message):
  Python bindings for channelizer.h are out of sync
Call Stack (most recent call first):
  python/bindings/CMakeLists.txt:40 (GR_PYBIND_MAKE_OOT)

-- Configuring incomplete, errors occurred!
```

eabase, 2024-11-06:
Apparently one should run something like:
```bash
# cd ..
#  gr_modtool bind channelizer

GNU Radio module name identified: lora
Please pass the compiler_path as argument to your xml_generator_configuration_t(), or add it to your pygccxml configuration file.
```

But it's missing something.

marcusmueller, 2024-11-06:
That's a problem with gr-lora having updated their channelizer.h, but not the hash in python/……/bindings/*channelizer.cc ; so, nothing that GNU Radio as framework can fix, you need to report this to gr-lora.

closed 2024-11-06 by marcusmueller

eabase, 2024-11-06:
Hi @marcusmueller 
Thanks, but I though the hash was generated by the `gr_modtool`?  
Only that the `gr_modtool` fails as shown above. Anyway, I see that this repo has not been updated for year and with many bugs and mixed Python2 and 3, so I have given up there.

marcusmueller, 2024-11-06:
note that at this year's GNU Radio conference (find the youtube links on https://grcon.stream ) there was a talk about a new better LoRA module. Don't remember which one that was.

eabase, 2024-11-07:
@marcusmueller 
I watched the video there, and although there were many interesting topics, there was no mentioning of any new/better LoRa module. In fact LoRa was not really mentioned at all, unless I missed something. So where did you see that?