gnuradio/gnuradio #7551: non-deterministic (flaky) qa_blocks_hier_block2 test failure

Apteryks, 2024-11-25:
### What happened?

The `qa_blocks_hier_block2` test has failed; with the only change being an updated meson and running the test in parallel.  It must be flaky.  See the Guix CI output: https://ci.guix.gnu.org/build/6607344/log, partially copied here:

```
starting phase `check'
Test project /tmp/guix-build-gnuradio-3.10.10.0.drv-0/build
_XSERVTransmkdir: ERROR: euid != 0,directory /tmp/.X11-unix will not be created.
        Start   1: runtime_qa_buffer.cc
        Start   2: runtime_qa_io_signature.cc
        Start   3: runtime_qa_logger.cc
        Start   4: runtime_qa_dictionary_logger.cc
        Start   5: runtime_qa_host_buffer.cc
        Start   6: runtime_qa_vmcircbuf.cc
        Start   7: math_qa_fxpt.cc
        Start   8: math_qa_fxpt_nco.cc
        Start   9: math_qa_fxpt_vco.cc
        Start  10: math_qa_math.cc
        Start  11: math_qa_sincos.cc
        Start  12: math_qa_fast_atan2f.cc
        Start  13: pmt_prims
        Start  14: pmt_unv
        Start  15: qa_flowgraph
        Start  16: qa_prefs
  1/260 Test   #2: runtime_qa_io_signature.cc ...................   Passed    0.03 sec
        Start  17: qa_python_logging
  2/260 Test   #3: runtime_qa_logger.cc .........................   Passed    0.03 sec
        Start  18: qa_random
  3/260 Test   #4: runtime_qa_dictionary_logger.cc ..............   Passed    0.03 sec
        Start  19: qa_sys_paths
  4/260 Test   #5: runtime_qa_host_buffer.cc ....................   Passed    0.04 sec
        Start  20: qa_tag_utils
  5/260 Test   #7: math_qa_fxpt.cc ..............................   Passed    0.03 sec
        Start  21: qa_hier_block2
  6/260 Test  #10: math_qa_math.cc ..............................   Passed    0.03 sec
        Start  22: qa_uncaught_exception
  7/260 Test  #11: math_qa_sincos.cc ............................   Passed    0.03 sec
        Start  23: qa_pmt
  8/260 Test  #12: math_qa_fast_atan2f.cc .......................   Passed    0.03 sec
        Start  24: qa_pmt_to_python
  9/260 Test  #14: pmt_unv ......................................   Passed    0.03 sec
        Start  25: blocks_qa_block_tags.cc
 10/260 Test  #25: blocks_qa_block_tags.cc ......................   Passed    0.04 sec
        Start  26: blocks_qa_gr_block.cc
 11/260 Test  #26: blocks_qa_gr_block.cc ........................   Passed    0.02 sec
        Start  27: blocks_qa_gr_flowgraph.cc
 12/260 Test  #27: blocks_qa_gr_flowgraph.cc ....................   Passed    0.02 sec
        Start  28: blocks_qa_gr_hier_block2.cc
 13/260 Test  #28: blocks_qa_gr_hier_block2.cc ..................   Passed    0.02 sec
        Start  29: blocks_qa_gr_hier_block2_derived.cc
 14/260 Test  #29: blocks_qa_gr_hier_block2_derived.cc ..........   Passed    0.02 sec
        Start  30: blocks_qa_gr_top_block.cc
 15/260 Test   #6: runtime_qa_vmcircbuf.cc ......................   Passed    0.18 sec
        Start  31: blocks_qa_rotator.cc
 16/260 Test   #9: math_qa_fxpt_vco.cc ..........................   Passed    0.17 sec
        Start  32: blocks_qa_set_msg_handler.cc
 17/260 Test   #8: math_qa_fxpt_nco.cc ..........................   Passed    0.18 sec
        Start  33: qa_add_mult_div_sub
 18/260 Test  #30: blocks_qa_gr_top_block.cc ....................   Passed    0.07 sec
        Start  34: qa_add_mult_v
 19/260 Test  #32: blocks_qa_set_msg_handler.cc .................   Passed    0.12 sec
        Start  35: qa_affinity
 20/260 Test  #31: blocks_qa_rotator.cc .........................   Passed    0.31 sec
        Start  36: qa_argmax
 21/260 Test  #23: qa_pmt .......................................   Passed    0.59 sec
        Start  37: qa_block_behavior
 22/260 Test  #15: qa_flowgraph .................................   Passed    0.62 sec
        Start  38: qa_block_gateway
 23/260 Test  #20: qa_tag_utils .................................   Passed    0.65 sec
        Start  39: qa_blockinterleaver_xx
[... 466 lines cut ...]
	 40 - qa_blocks_hier_block2 (Failed)
Errors while running CTest
error: in phase 'check': uncaught exception:
%exception #<&invoke-error program: "ctest" arguments: ("-j" "16" "--output-on-failure" "-E" "qa_header_payload_demux|qa_packet_headerparser_b|qa_rotator_cc") exit-status: 8 term-signal: #f stop-signal: #f> 
phase `check' failed after 41.5 seconds
command "ctest" "-j" "16" "--output-on-failure" "-E" "qa_header_payload_demux|qa_packet_headerparser_b|qa_rotator_cc" failed with status 8
build process 18 exited with status 256
```

### System Information

OS: Guix System
GR Installation Method: Source/Guix

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.10.0

### Steps to Reproduce the Problem

`guix build gnuradio` to build it once/fetch a substitute, then `guix build gnuradio --no-grafts --rounds=100` to build it many times until it fails on an up-to-date guix should do it.

Apteryks, 2024-11-25:
Hm, reproducing this will be tricky even with Guix, as the `gnuradio` package doesn't build reproducibly yet and `guix build gnuradio --no-grafts --rounds=100` fails with `guix build: error: derivation `/gnu/store/d4ira96pnn2kayi3rdwpk23ssb92gi4z-gnuradio-3.10.10.0.drv' may not be deterministic: output `/gnu/store/f94xk07s9z4nrjhv8f5d7qj477yy8g21-gnuradio-3.10.10.0' differs from `/gnu/store/f94xk07s9z4nrjhv8f5d7qj477yy8g21-gnuradio-3.10.10.0-check'`.  I've reported that non-reproducibility here: https://issues.guix.gnu.org/74521

marcusmueller, 2024-11-25:
This is going to be hard to debug without a backtrace and core dump! Could you try enable core dumps for segfaulting processes on your system, and try to figure out where exactly the segfault happens?

marcusmueller, 2024-11-25:
That would be very appreciated, especially feedback arising from the process of getting things to build reproducibly.

I'd tend to let you decide whether to close this issue or keep it open; you're the one of us with the best assessment whether it might be a persistent bug. Proposal: we leave it open, but I ping you in here on 2025-01-14, and if it hasn't happen till then, we just close it, unless you think it was a fluke caused by something on your system and we close it right now. If you find any indication that it still happens, we of course keep the issue open?

marcusmueller, 2025-01-14:
@Apteryks any news? Sadly, it might be we're seeing the same problem elsewhere: https://github.com/gnuradio/gnuradio/pull/7656

Apteryks, 2025-01-16:
> @Apteryks any news? Sadly, it might be we're seeing the same problem elsewhere: #7656

Hi Marcus!  Sorry, no!  Feel free to close; I can reopen the next time I encounter it, if I could collect more useful  data about the failure.

marcusmueller, 2025-01-16:
Nah, now that I saw that sporadically happen, there's no reason to close this

argilo, 2025-02-02:
I ran the test in a loop with core dumps enabled, and observed a few different backtraces:

```
#0  std::__uniq_ptr_impl<boost::thread, std::default_delete<boost::thread> >::_M_ptr (this=0x40deb0c3) at /usr/include/c++/13/bits/unique_ptr.h:199
#1  std::unique_ptr<boost::thread, std::default_delete<boost::thread> >::get (this=0x40deb0c3) at /usr/include/c++/13/bits/unique_ptr.h:470
#2  std::unique_ptr<boost::thread, std::default_delete<boost::thread> >::operator-> (this=0x40deb0c3) at /usr/include/c++/13/bits/unique_ptr.h:463
#3  gr::thread::thread_group::join_all (this=0x40daf798) at /home/argilo/prefix_311/src/gnuradio/gnuradio-runtime/lib/thread/thread_group.cc:62
#4  0x000072dde0378eab in gr::top_block_impl::wait (this=0x40db2910) at /home/argilo/prefix_311/src/gnuradio/gnuradio-runtime/lib/top_block_impl.cc:140
#5  0x000072dde036ef90 in gr::top_block::wait (this=<optimized out>) at /home/argilo/prefix_311/src/gnuradio/gnuradio-runtime/lib/top_block.cc:64
#6  0x000072dde8d755de in top_block_wait_unlocked (r=std::shared_ptr<gr::top_block> (use count 3, weak count 1) = {...}) at /usr/include/c++/13/bits/shared_ptr_base.h:1665
#7  0x000072dde8d7b41a in pybind11::detail::argument_loader<std::shared_ptr<gr::top_block> >::call_impl<void, void (*&)(std::shared_ptr<gr::top_block>), 0ul, pybind11::detail::void_type>(void (*&)(std::shared_ptr<gr::top_block>), std::integer_sequence<unsigned long, 0ul>, pybind11::detail::void_type&&) && (f=<optimized out>, this=0x72dd93dff7d0) at /usr/include/pybind11/cast.h:1480
#8  pybind11::detail::argument_loader<std::shared_ptr<gr::top_block> >::call<void, pybind11::detail::void_type, void (*&)(std::shared_ptr<gr::top_block>)>(void (*&)(std::shared_ptr<gr::top_block>)) && (f=<optimized out>, this=0x72dd93dff7d0) at /usr/include/pybind11/cast.h:1454
#9  pybind11::cpp_function::initialize<void (*&)(std::shared_ptr<gr::top_block>), void, std::shared_ptr<gr::top_block>, pybind11::name, pybind11::scope, pybind11::sibling>(void (*&)(std::shared_ptr<gr::top_block>), void (*)(std::shared_ptr<gr::top_block>), pybind11::name const&, pybind11::scope const&, pybind11::sibling const&)::{lambda(pybind11::detail::function_call&)#1}::operator()(pybind11::detail::function_call&) const (call=..., __closure=0x0) at /usr/include/pybind11/pybind11.h:254
#10 pybind11::cpp_function::initialize<void (*&)(std::shared_ptr<gr::top_block>), void, std::shared_ptr<gr::top_block>, pybind11::name, pybind11::scope, pybind11::sibling>(void (*&)(std::shared_ptr<gr::top_block>), void (*)(std::shared_ptr<gr::top_block>), pybind11::name const&, pybind11::scope const&, pybind11::sibling const&)::{lambda(pybind11::detail::function_call&)#1}::_FUN(pybind11::detail::function_call&) () at /usr/include/pybind11/pybind11.h:224
#11 0x000072dde8c8d787 in pybind11::cpp_function::dispatcher (self=<optimized out>, args_in=(<gnuradio.gr.gr_python.top_block_pb at remote 0x72de0b52a270>,), kwargs_in=0x0)
    at /usr/include/pybind11/pybind11.h:946
#12 0x000000000058238f in cfunction_call (func=<built-in method top_block_wait_unlocked of PyCapsule object at remote 0x72ddf171ac40>, args=<optimized out>, kwargs=<optimized out>)
    at ../Objects/methodobject.c:537
#13 0x0000000000549225 in _PyObject_MakeTpCall (tstate=0x40da53b0, callable=<built-in method top_block_wait_unlocked of PyCapsule object at remote 0x72ddf171ac40>, args=<optimized out>, nargs=1, 
    keywords=0x0) at ../Objects/call.c:240
#14 0x0000000000549c5d in _PyObject_VectorcallTstate (kwnames=<optimized out>, nargsf=<optimized out>, args=<optimized out>, callable=<optimized out>, tstate=<optimized out>)
    at ../Include/internal/pycore_call.h:90
#15 0x00000000005d7909 in _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at Python/bytecodes.c:2706
#16 0x000000000054cde2 in _PyObject_VectorcallTstate (kwnames=0x0, nargsf=1, args=0x72dd93dffe18, callable=<function at remote 0x72de0f176c00>, tstate=0x40da53b0)
    at ../Include/internal/pycore_call.h:92
#17 method_vectorcall (method=<optimized out>, args=0xb486c0 <_PyRuntime+76288>, nargsf=<optimized out>, kwnames=0x0) at ../Objects/classobject.c:69
--Type <RET> for more, q to quit, c to continue without paging--
#18 0x00000000006f8adc in thread_run (boot_raw=0x408d3970) at ../Modules/_threadmodule.c:1114
#19 0x00000000006b98cc in pythread_wrapper (arg=<optimized out>) at ../Python/thread_pthread.h:237
#20 0x000072de0f89ca94 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:447
#21 0x000072de0f929c3c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

```
#0  0x000074826f06a564 in boost::thread::get_thread_info() const () from /lib/x86_64-linux-gnu/libboost_thread.so.1.83.0
#1  0x000074826f06d2ba in boost::thread::native_handle() () from /lib/x86_64-linux-gnu/libboost_thread.so.1.83.0
#2  0x000074824c9b7c8c in boost::thread::get_id (this=0x100000001) at /usr/include/boost/thread/detail/thread.hpp:738
#3  boost::thread::join (this=0x100000001) at /usr/include/boost/thread/detail/thread.hpp:759
#4  gr::thread::thread_group::join_all (this=0x6005628) at /home/argilo/prefix_311/src/gnuradio/gnuradio-runtime/lib/thread/thread_group.cc:62
#5  0x000074824c978eab in gr::top_block_impl::wait (this=0x6005860) at /home/argilo/prefix_311/src/gnuradio/gnuradio-runtime/lib/top_block_impl.cc:140
#6  0x000074824c96ef90 in gr::top_block::wait (this=<optimized out>) at /home/argilo/prefix_311/src/gnuradio/gnuradio-runtime/lib/top_block.cc:64
[... cut ...]
#24 std::__invoke_r<void, gr::thread::thread_body_wrapper<gr::tpb_container>&> (__fn=...) at /usr/include/c++/13/bits/invoke.h:111
#25 std::_Function_handler<void (), gr::thread::thread_body_wrapper<gr::tpb_container> >::_M_invoke(std::_Any_data const&) (__functor=...) at /usr/include/c++/13/bits/std_function.h:290
#26 0x00007c045704d99f in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.83.0
#27 0x00007c045be9ca94 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:447
#28 0x00007c045bf29c3c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

argilo, 2025-02-03:
The specific test that triggers a segfault is this one:

https://github.com/gnuradio/gnuradio/blob/efe5c8680b4ee524fc54267e7abed25a4977c886/gr-blocks/python/blocks/qa_blocks_hier_block2.py#L455-L477

argilo, 2025-02-03:
I believe the issue is this:

* Both `wait()` (called from the test's main thread) and `unlock()` (called from a background thread) ultimately end up calling `join()` to wait for scheduler threads to terminate.
* Concurrently calling `join()` on the same thread object from multiple threads constitutes a data race that results in undefined behavior.

The concurrent calls only occur when `unlock()` is called rapidly after `lock()`, and before the scheduler threads have had a chance to terminate.

It appears this issue was previously reported in #7276.

argilo, 2025-02-03:
With a bit of debug logging it can be observed that undefined behaviour indeed occurs as the test executes:
```diff
diff --git a/gnuradio-runtime/lib/top_block_impl.cc b/gnuradio-runtime/lib/top_block_impl.cc
index 4580d0e866..921839e627 100644
--- a/gnuradio-runtime/lib/top_block_impl.cc
+++ b/gnuradio-runtime/lib/top_block_impl.cc
@@ -155,8 +155,11 @@ void top_block_impl::wait()
 
 void top_block_impl::wait_for_jobs()
 {
-    if (d_scheduler)
+    if (d_scheduler) {
+        std::cout << "wait_for_jobs      " << std::this_thread::get_id() << " " << d_scheduler.get() << std::endl;
         d_scheduler->wait();
+        std::cout << "done wait_for_jobs " << std::this_thread::get_id() << " " << d_scheduler.get() << std::endl;
+    }
 }
 
 // N.B. lock() and unlock() cannot be called from a flow graph
```

Lines 1-4 below show the concurrent `wait()` calls on the same scheduler; these call `join` under the covers.
```
wait_for_jobs      137434746586816 0x3115ead0
wait_for_jobs      137434736101056 0x3115ead0
done wait_for_jobs 137434746586816 0x3115ead0
done wait_for_jobs 137434736101056 0x3115ead0
wait_for_jobs      137434736101056 0x3115b820
done wait_for_jobs 137434736101056 0x3115b820
wait_for_jobs      137436755562816 0x3115b820
done wait_for_jobs 137436755562816 0x3115b820
```

argilo, 2025-02-03:
It would appear that the `top_block` methods are intended to be thread-safe, because they use a mutex to prevent concurrent access to internal state. So they should probably use a mechanism other then `join` to determine when the scheduler has terminated. (A condition variable, perhaps?)

argilo, 2025-02-03:
Calling `wait()` from multiple threads also triggers the same undefined behaviour. This program reliably crashes within seconds on my laptop:

```python
#!/usr/bin/env python3

import threading
from gnuradio import gr, blocks

while True:
    tb = gr.top_block()
    src = blocks.vector_source_f([0]*3200)
    throttle = blocks.throttle(gr.sizeof_float, 32000)
    sink = blocks.null_sink(gr.sizeof_float)
    tb.connect(src, throttle, sink)
    tb.start()

    def my_thread(tb):
        tb.wait()

    for _ in range(50):
        threading.Thread(target=my_thread, args=(tb, )).start()

    tb.wait()
```

willcode, 2025-02-03:
Nice debugging! I was going down the path of concurrent thread operations but didn't notice what the unlock did.