gnuradio/gnuradio #7558: Circular Dependency in `gnuradio/grc/core`

sreevatsank1999, 2024-11-29:
### What happened?

`gr_modtool` failed due to a circular dependency in `grc/core`. 

### System Information

OS: Ubuntu 22.04
GR Installation Method: From Source (v3.10.9.1)

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.1

### Steps to Reproduce the Problem

Run `gr_modtool update --complete` in a gr OOT module directory.

Note: the bug is not specific to `gr_modtool`. 

### Relevant log output

```shell
Traceback (most recent call last):
  File "/usr/local/bin/gr_modtool", line 18, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/modtool/cli/update.py", line 32, in cli
    run(self)
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/modtool/cli/base.py", line 155, in run
    module.run()
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/modtool/core/update.py", line 61, in run
    from gnuradio.grc.converter import Converter
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/grc/converter/__init__.py", line 8, in <module>
    from .main import Converter
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/grc/converter/main.py", line 13, in <module>
    from . import block_tree, block
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/grc/converter/block_tree.py", line 11, in <module>
    from ..core.io import yaml
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/grc/core/io/yaml.py", line 12, in <module>
    from ..params.param import attributed_str
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/grc/core/params/__init__.py", line 7, in <module>
    from .param import Param
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/grc/core/params/param.py", line 17, in <module>
    from . import dtypes
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/grc/core/params/dtypes.py", line 13, in <module>
    from .. import blocks
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/grc/core/blocks/__init__.py", line 11, in <module>
    from ..params import Param
ImportError: cannot import name 'Param' from partially initialized module 'gnuradio.grc.core.params' (most likely due to a circular import) (/usr/local/lib/python3.10/dist-packages/gnuradio/grc/core/params/__init__.py)
```

marcusmueller, 2024-12-14:
@sreevatsank1999 thanks! Do you happen to know if the code is the same on the `main` branch? If so, would you be able to open a pull request?

ffu, 2025-01-15:
I've also seen this error, and it can be reproduced by entering the following:
```
import gnuradio.grc.core.params
```
One possible solution is to modify core/blocks/blocks.py to import the params package instead of the Param class. That means changing the import statement to "import ..params" instead of "from ..params import Param". The Param class is used only once in blocks.py, and it's for a type hint, so the fix may only require two line changes. When I perform the change, I no longer see ImportError.

willcode, 2025-01-15:
Use this: `from gnuradio.grc.platform import params`

ffu, 2025-01-24:
Yes, as long as the gnuradio.grc.core.blocks package is imported before the params package, there's no error. The platform module does exactly that.

ffu, 2025-04-23:
I believe the circular dependency issue has been fixed in PR #7772 
Previously, the gnuradio.grc.core.params package could not be imported without gnuradio.grc.core.blocks being imported first. This ordering is not intuitive and can easily create exceptions as seen in this issue. 

The PR referenced above included changes in core/blocks/block.py that deleted the import of the core.params.Param class. This removes the circular dependency between the core.blocks package and the core.params package. After the change, one can import gnuradio.grc.core.params first without any errors.