gnuradio/gnuradio #7562: In the MSYS2 environment of Windows, adding the oot module results in a runtime error

CP3164094, 2024-12-12:
### What happened?

When I input ninja, the error is as follows
[image]
[image]
[image]

### System Information

OS: ( Windows)

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.11.0

### Steps to Reproduce the Problem

cmake -DPYTHON_EXECUTABLE="${MINGW_PREFIX}"/bin/python -DCMAKE_INSTALL_PREFIX="${MINGW_PREFIX}" ..
[image]
ninja
and then error as shown in the above figure
what should i do
Thanks

marcusmueller, 2024-12-12:
Hi, as @chrisjohgorman said, it's better to post text than screenshots of text; we can't search for text in images.

chrisjohgorman, 2024-12-12:
Hi @CP3164094, I'm willing to try to help you debug this, but would need a minimal failing example, or zip file (or tar ball) of your code to do much more.  I've tried both tutorials on making a gr-module and they build without error.  I also tried a bunch of other gr-modules (radar, osmosdr, correctiq and m2k) and none are showing your error.

marcusmueller, 2024-12-14:
added the "clarification needed" label. Will close issue at the beginning of next year if we can't reproduce by then; also, still insufficient to just have screenshots of console text instead of text.

CP3164094, 2024-12-16:
> Hi @CP3164094, I'm willing to try to help you debug this, but would need a minimal failing example, or zip file (or tar ball) of your code to do much more. I've tried both tutorials on making a [...]

Sorry, it was my code issue and it just showed up like this. I have already resolved it. Thank you for your reply.

closed 2024-12-16 by CP3164094