gnuradio/gnuradio #7563: FEC Extended Encoder / CC Decoder Definition, Segmentation Fault, Likely bug in Volk

SJ-Innovation, 2024-12-12:
### What happened?

Initially reported in. 
https://github.com/daniestevez/gr-satellites/issues/692

Direct usage of (or hier instanciation of) the FEC Extended Decoder with a CC Decoder definition produces a segfault on startup of the flowgraph. This has been traced to a likely bug in Volk or its usage as ```VOLK_GENERIC=1``` works around the issue. This is a Windows specific issue, as both Mac M1 (CB) & Arch Linux (DE) do not exhibit the same behaviour.

### System Information

OS: Windows 10
GR: 3.10.11.0, Py 3.12.5
GR Installation Method: (Ryanvolz/RadioConda)

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.11.0

### Steps to Reproduce the Problem

Any flowgraph using the FEC Extended Decoder along with a CC Decoder definition (screenshotted) produces a segmentation fault. 
Setting ```VOLK_GENERIC=1``` works around the issue, though obviously performance takes a hit. This suggests theres a bug in Volk, or in its usage by the decoder. 
[image]

SJ-Innovation, 2024-12-12:
I've just been doing this. Also interestingly this kernel seems to have come up as a bugbear in the past:  https://github.com/gnuradio/volk/issues/395

Of the 2 options I have "spiral" and "generic" only the generic one seems to run. Spiral gives a segfault. Its also probably worth nothing that spiral is 10x faster than generic so preferable to maintain the speed, though thats not worth anything if its broken!

I've switched volk_8u_x4_conv_k7_r2_8u kernel over to generic and it seems to be working, so i think thats nailed it down to the spiral volk_8u_x4_conv_k7_r2_8u kernel.

marcusmueller, 2024-12-12:
Assigning this to @jdemel, he's most likely to have insight on that specific kernel.

Also, I've been long not amazed by the fact we have non-vector kernels hiding in a library called "vector library of kernels", and that these kernels are the main source of crashes.

@daniestevez @noc0lour is this specific decoder something we can just get from aff3ct?

daniestevez, 2024-12-12:
I think aff3ct doesn't have a Viterbi decoder for the CCSDS r=1/2 k=7 convolutional code, which is what we want here. Also, honestly, since helping with the aff3ct GSoC this year, I'm a bit more disappointed with the state of aff3ct. To me it looks like an awesome tool to do simulations for BER plots, but using it as a library is a pain, and it's not particularly fast for production use either.

I don't agree that this is a non-vector kernel. It calculates multiple butterflies in parallel using SIMD. I understand that a Viterbi decoder is far more specialized than, say, a sum of two vectors, and understand the concerns about whether the scope of Volk should only be math kernels or also FEC. But it's also hard to draw a strict line. CPUs these days even include SIMD instructions for finite field polynomial arithmetic (just to give an example of something which is not simple math), and between a vector sum an a Viterbi decoder, there are potentially a myriad of kernels of different complexity which may or may not fit in Volk.

Also, if I had to take my guess, I would say that none of the Viterbi kernels in Volk are to blame for these bugs (specially the Spiral one, which is machine generated). My guess is that the bugs lie in how the kernels are called from gr-fec. Example of such bug: https://github.com/gnuradio/gnuradio/pull/6445 Maybe there are more like this lurking. I only found this bug because it was very rarely causing bad decodes in frames of good enough SNR.

marcusmueller, 2024-12-15:
hey, sorry for the slightly ranty/bad wording approach there: What I meant is that kernels carrying what I'd call a "complex persistent state" are really not a great fit, and this particular one I find problematic because it never had documentation for the parameters it uses (which is due to these being taken and named from a pure-C implementation, probably by Phil Karn?), with the effect that every bug fix involves significant reverse engineering :)

I'll be a bit careful about saying `volk_78_x4_conv_k7_r2_8u_spiral` (i.e., the original SSE3 kernel) is generated by spiral; it was, indeed, then adapted to a VOLK signature. If I recall history (with a bit of help from `git log`) correctly: We then went and "transcribed" the SSE3 code to AVX2, which then consequently broke and was removed, we then basically repeated that experiment and transcribed the SSE3 kernel to NEON using `sse2neon`, and finally Clayton cleaned up a lot of the spiral-derived code to make it readable and undo modern-compiler antipatterns like manually unrolled loops, and fixed the AVX2 kernel.

You're right for calling me out on incorrectly saying that this kernel isn't vectorized! It massively is. The thing I should have complained about is that we're noticing all the bugs of our decoder kernels in application, not in testing, because the VOLK testing is suited for testing "memoryless, vector-in-vector-out" kernels, not for catching decoders that perform half as good as they should; or at least, we're really not doing that in our puppets.