gnuradio/gnuradio #7564: gnuradio-3.10.11.0 fails to compile with boost >= 1.87.0

ChrisVine, 2024-12-14:
### What happened?

With boost-1.87.0, compilation of gnuradio-3.10.11.0 fails at gr-network/lib/socket_pdu_impl.cc with error message:

```socket_pdu_impl.cc:57:34: error: ‘query’ is not a member of ‘boost::asio::ip::tcp::resolver’ {aka ‘boost::asio::ip::basic_resolver<boost::asio::ip::tcp>’}```

asio::ip::tcp::resolver:query was deprecated (together with some other boost::asio functionality which may or may not be used by gnuradio) about 7 years' ago at boost-1.66, and seems now to have been removed.

### System Information

Linux

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.11.0

### Steps to Reproduce the Problem

Attempt to compile gnuradio-3.10.11.0 from source.

marcusmueller, 2024-12-14:
solved by https://github.com/gnuradio/gnuradio/pull/7554

marcusmueller, 2024-12-14:
@ChrisVine It looks like we have a fix for that in the pipeline, see the issue linked above. Maybe you could try whether the fix works for you!

ChrisVine, 2024-12-15:
I have applied the fix and gnuradio now compiles and runs fine with boost-1.87, but I may not be the best as my flowgraph does not knowingly use gr-network.

closed 2024-12-15 by ChrisVine