gnuradio/gnuradio #7566: test failure for qa_correlate_access_code_XX_ts.py

ZeroChaos-, 2024-12-17:
### What happened?

```
181/267 Test #192: qa_correlate_access_code_XX_ts ...............***Failed    0.34 sec
......E
======================================================================
ERROR: test_payload_lengths (__main__.test_correlate_access_code_XX_ts.test_payload_lengths)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999/gr-digital/python/digital/qa_correlate_access_code_XX_ts.py", line 104, in test_payload_lengths
    self._variable_payload_t(length)
  File "/var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999/gr-digital/python/digital/qa_correlate_access_code_XX_ts.py", line 77, in _variable_payload_t
    self.assertEqual(header[0] * 256 + header[1], header[2] * 256 + header[3])
                     ~~~~~~~~~~^~~~~
OverflowError: Python integer 256 out of bounds for uint8
```

### System Information

OS:  Gentoo
GR Installation Method: Native distro package (compiling with package manager help)

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

v3.10.11.0-1-gee27d6f3

### Steps to Reproduce the Problem

Build and run tests

### Relevant log output

```shell
>>> Emerging (1 of 1) net-wireless/gnuradio-9999::maintainer-zero
 * Using python3.12 to build
>>> Unpacking source...
 * Repository id: gnuradio_gnuradio.git
 * To override fetched repository properties, use:
 *   EGIT_OVERRIDE_REPO_GNURADIO_GNURADIO
 *   EGIT_OVERRIDE_BRANCH_GNURADIO_GNURADIO
 *   EGIT_OVERRIDE_COMMIT_GNURADIO_GNURADIO
 *   EGIT_OVERRIDE_COMMIT_DATE_GNURADIO_GNURADIO
 * 
 * Fetching https://github.com/gnuradio/gnuradio.git ...
git fetch https://github.com/gnuradio/gnuradio.git +refs/heads/maint-3.10:refs/heads/maint-3.10
git symbolic-ref refs/git-r3/net-wireless/gnuradio/0/__main__ refs/heads/maint-3.10
 * Checking out https://github.com/gnuradio/gnuradio.git to /var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999 ...
git checkout --quiet maint-3.10
GIT update -->
   repository:               https://github.com/gnuradio/gnuradio.git
   at the commit:            ee27d6f3d63b39e1427f80de9950d9a64bdae096
>>> Source unpacked in /var/tmp/portage/net-wireless/gnuradio-9999/work
>>> Preparing source in /var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999 ...
 * Source directory (CMAKE_USE_DIR): "/var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999"
 * Build directory  (BUILD_DIR):     "/var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999_build"
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999 ...
 * Source directory (CMAKE_USE_DIR): "/var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999"
 * Build directory  (BUILD_DIR):     "/var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999_build"
cmake -C /var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DENABLE_DEFAULT=OFF -DENABLE_GNURADIO_RUNTIME=ON -DENABLE_PYTHON=ON -DENABLE_GR_BLOCKS=ON -DENABLE_GR_ANALOG=ON -DENABLE_GR_AUDIO=ON -DENABLE_GR_CHANNELS=ON -DENABLE_GR_CTRLPORT=OFF -DENABLE_GR_DIGITAL=ON -DENABLE_DOXYGEN=OFF -DENABLE_GR_DTV=ON -DENABLE_GR_FEC=ON -DENABLE_GR_FFT=ON -DENABLE_GR_FILTER=ON -DENABLE_GRC=ON -DENABLE_GR_IIO=ON -DENABLE_GR_MODTOOL=ON -DENABLE_GR_BLOCKTOOL=ON -DENABLE_GR_NETWORK=ON -DENABLE_GR_PDU=ON -DENABLE_PERFORMANCE_COUNTERS=OFF -DENABLE_TESTING=ON -DENABLE_GR_QTGUI=ON -DENABLE_GR_SOAPY=ON -DENABLE_GR_TRELLIS=ON -DENABLE_GR_UHD=ON -DENABLE_GR_UTILS=ON -DENABLE_GR_VIDEO_SDL=ON -DENABLE_GR_VOCODER=ON -DENABLE_GR_WAVELET=ON -DENABLE_GR_ZEROMQ=ON -DSYSCONFDIR=/etc -DPYTHON_EXECUTABLE=/usr/bin/python3.12 -DGR_PYTHON_DIR=/usr/lib/python3.12/site-packages -DGR_PKG_DOC_DIR=/usr/share/doc/gnuradio-9999 -DMATHJAX2_ROOT=/usr/share/mathjax -DCMAKE_BUILD_TYPE=None -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999_build/gentoo_toolchain.cmake /var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999
loading initial cache file /var/tmp/portage/net-wireless/gnuradio-9999/work/gnuradio-9999_build/gentoo_common_config.cmake
-- The CXX compiler identification is GNU 14.2.1
-- The C compiler identification is GNU 14.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++ - skipped
[... 509 lines cut ...]
--   * post-install
--   * doxygen
--   * man-pages
--   * common-precompiled-headers
--   * gr-ctrlport
--   * JSON/YAML config blocks
--   * gr-uhd UHD 4.0 RFNoC
-- 
-- Using install prefix: /usr
-- Building for version: v3.10.11.0-1-gee27d6f3 / 3.10.12git
-- <<< Gentoo configuration >>>
Build type      None
Install path    /usr
Compiler flags:
C               -mtune=core2 -pipe -O3 -flto -frecord-gcc-switches -Wstringop-overread -Wformat -Wformat-security -D_FORTIFY_SOURCE=3 -D_GLIBCXX_ASSERTIONS -ftrivial-auto-var-init=pattern -Wl,-z,relro,-z,now -fstack-protector-strong -fstack-clash-protection -fcf-protection=full -Werror=strict-aliasing -Werror=odr -Werror=lto-type-mismatch -Werror=stringop-overread -Werror=format-security -march=alderlake -mabm -mno-cldemote -mno-kl -mno-sgx -mno-widekl -mshstk --param=l1-cache-line-size=64 --param=l1-cache-size=32 --param=l2-cache-size=36864 -fPIE -pie -ggdb3 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wignored-qualifiers -Wcast-qual
C++             -mtune=core2 -pipe -O3 -flto -frecord-gcc-switches -Wstringop-overread -Wformat -Wformat-security -D_FORTIFY_SOURCE=3 -D_GLIBCXX_ASSERTIONS -ftrivial-auto-var-init=pattern -Wl,-z,relro,-z,now -fstack-protector-strong -fstack-clash-protection -fcf-protection=full -Werror=strict-aliasing  -Werror=lto-type-mismatch -Werror=stringop-overread -Werror=format-security -march=alderlake -mabm -mno-cldemote -mno-kl -mno-sgx -mno-widekl -mshstk --param=l1-cache-line-size=64 --param=l1-cache-size=32 --param=l2-cache-size=36864 -fPIE -pie -ggdb3 -fPIC -fPIC -fPIC -fPIC -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wignored-qualifiers -Wcast-qual -fcx-limited-range
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0
Module          -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0
Shared          -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0

-- Configuring done (19.7s)
-- Generating done (0.4s)
```
```

ZeroChaos-, 2024-12-17:
This is running in Gentoo's installed python 3.12
```
zero@naga gnuradio % python --version                                                                                                                                                (git)-[master]  1 ⨯
Python 3.12.8
```

marcusmueller, 2024-12-19:
I introduced that bug, but Argilo already fixed it on main in 7d61746

marcusmueller, 2024-12-19:
Looks like we just need to finish backporting of #7499

marcusmueller, 2024-12-20:
closed via #7570 . For some reason, the magic close verb on that PR didn't close this automagically.

closed 2024-12-20 by marcusmueller