gnuradio/gnuradio #7578: Tests involving add_block_from_query fail due to query not producing a block

marcusmueller, 2024-12-26:
Extracted from #6957.

marcusmueller, 2024-12-26:
@AsriFox  said:

The problem with `add_block_from_query` looks like this:
[image]
Here are the test logs from running `ctest -R grc_tests` with `QT_QPA_PLATFORM=wayland` and `QT_QPA_PLATFORM=xcb` on different systems:
[Arch_Hyprland_wayland_Test.log](https://github.com/gnuradio/gnuradio/files/13550657/Arch_Hyprland_wayland_Test.log)
[Arch_Hyprland_xcb_Test.log](https://github.com/gnuradio/gnuradio/files/13550658/Arch_Hyprland_xcb_Test.log)
[Fedora_Gnome45.2_wayland_Test.log](https://github.com/gnuradio/gnuradio/files/13550659/Fedora_Gnome45.2_wayland_Test.log)
[Fedora_Gnome45.2_xcb_Test.log](https://github.com/gnuradio/gnuradio/files/13550660/Fedora_Gnome45.2_xcb_Test.log)
[Ubuntu_Gnome42.9_wayland_Test.log](https://github.com/gnuradio/gnuradio/files/13550661/Ubuntu_Gnome42.9_wayland_Test.log)
[Ubuntu_Gnome42.9_xcb_Test.log](https://github.com/gnuradio/gnuradio/files/13550663/Ubuntu_Gnome42.9_xcb_Test.log)
[Ubuntu_Gnome45_xcb_Test.log](https://github.com/gnuradio/gnuradio/files/13550664/Ubuntu_Gnome45_xcb_Test.log)
[Ubuntu_Plasma_xcb_Test.log](https://github.com/gnuradio/gnuradio/files/13550665/Ubuntu_Plasma_xcb_Test.log)

marcusmueller, 2024-12-30:
don't have a wayland system at hand, hence can't try to repro

Anjali503, 2026-01-04:
I can try to reproduce this on my setup and report back with details.

I don’t have Wayland readily available, but I can run the failing grc_tests_add_block_from_query tests on XCB and inspect what queries fail to resolve into blocks, and whether the block registry differs between environments.

I’ll comment with logs or observations once I have something concrete.